moveit/panda_moveit_config #6: Discrepancy with Fanka Emika's panda_moveit_config

christgau, 2018-06-29:
I'm struggling with the panda moveit configuration. In the simulation, I use the latest version from panda_moveit_config since it solves the problems with the gripper I encountered some time ago (see https://github.com/ros-planning/moveit/issues/948#issuecomment-397569356 for details). So, simulations run fine: simple motion, picking up objects and placing just works.

However, when it comes to the real robot, I run into some problems. First, according to https://frankaemika.github.io/docs/ros_introduction.html and https://github.com/frankaemika/franka_ros running `roslaunch panda_moveit_config panda_moveit.launch` should load the MoveIt configuration for the Panda. But, in the ros_planning version there is no panda_moveit.launch file. Is this intended? Is move_group.launch a replacement although it does not load the `position_joint_trajectory_controller` as Emika's panda_moveit.launch file does?

Or is the configuration provided by the ros-planning/panda_moveit_config solely intended for simulation? (a clue to this is that package.xml does not list franka_description as runtime dependency) 

Nevertheless, there are further differences (see this [gist](https://gist.github.com/christgau/7b4ddde69d6d7d845ba7b177d421806d)). For example, the planning groups are different. Franka Emika provides panda_arm_hand while ros_planning does not. I'm confused which version one should use for development. My assumption is that if code runs fine in simulation it should also do with a real robot. But that's unfortunately not the case.

christgau, 2018-07-02:
Here are some more details from a session with the real robot

panda_moveit_config is replaced with the latest version from Github (see OP). I first load/run franka_control via franka_control.launch with the robot IP. Then, I launch panda_moveit_config with the moveit_group launch file and after that I start my ROS node which uses moveit to steer the robot.

Settings a pose goal then results in the following
```
...
[ INFO] [1530542406.659806254]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1530542406.659867536]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1530542406.659891438]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1530542406.659901681]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1530542406.659921995]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1530542406.659944098]: Using planning request adapter 'Fix Start State Path Constraints'
[ WARN] [1530542411.674359780]: Waiting for panda_arm_controller/follow_joint_trajectory to come up
[ WARN] [1530542417.674537451]: Waiting for panda_arm_controller/follow_joint_trajectory to come up
[ERROR] [1530542423.674869664]: Action client not connected: panda_arm_controller/follow_joint_trajectory
[ WARN] [1530542428.705509854]: Waiting for hand_controller/gripper_action to come up
[ WARN] [1530542434.705747788]: Waiting for hand_controller/gripper_action to come up
[ERROR] [1530542440.706103485]: Action client not connected: hand_controller/gripper_action
[ INFO] [1530542440.726892955]: Returned 0 controllers in list
[ INFO] [1530542440.737690492]: Trajectory execution is managing controllers

...

You can start planning now!

[ INFO] [1530542771.884714952]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1530542771.884992501]: Planning attempt 1 of at most 1
[ INFO] [1530542771.885872111]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1530542771.886201932]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1530542771.947844346]: RRTConnect: Created 5 states (3 start + 2 goal)
[ INFO] [1530542771.947878331]: Solution found in 0.061747 seconds
[ INFO] [1530542771.955166692]: SimpleSetup: Path simplification took 0.007182 seconds and changed from 4 to 2 states
[ INFO] [1530542771.956842235]: Returned 0 controllers in list
[ERROR] [1530542771.956865558]: Unable to identify any set of controllers that can actuate the specified joints: [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 ]
[ERROR] [1530542771.956903312]: Known controllers and their joints:

[ERROR] [1530542771.956927919]: Apparently trajectory initialization failed
```

krigi100, 2018-07-25:
Have you already found a Solution?
I have the same Problem. I can't use Pick and Place with the real Robot, with the Simulated Robot it's possible.

christgau, 2018-07-26:
> Have you already found a Solution?

Actually, yes. I've ben able to solve the issue, by changing some configuration files. That's the good news. I really would like to share my solution with the community but - and thats the bad news - I'm still in discussion with the legal department and processes take a looooong time. However, the fix is rather small.

krigi100, 2018-07-26:
Ok thanks,
i found a dirty, temporarily solution by copy the [panda.srdf](https://github.com/ros-planning/panda_moveit_config/blob/master/config/panda.srdf) in the [franka_ros/panda_moveit_config/config](https://github.com/frankaemika/franka_ros/tree/kinetic-devel/panda_moveit_config/config) and use the file in the [planning_context.launch](https://github.com/frankaemika/franka_ros/blob/kinetic-devel/panda_moveit_config/launch/planning_context.launch) file.

fwalch, 2018-08-10:
@christgau If the legal process is still ongoing, can you maybe share the cause of the problem instead of the fix? We're trying to merge the two panda_moveit_configs (see #9), but that would probably introduce the bug into this repo as well.

christgau, 2018-08-10:
@fwalch As explained in the issue, one thing is the existence of  different planning groups, i.e. the extra `panda_arm_hand` group in the franka_ros version of panda_moveit_config. This causes some confusion. In the ros_planning version, the group does not exist and it seems not to be required. It appears to me, that this planning group is of no use. But I may be mistaken.

In addition, I needed to create the `panda_moveit.launch` file which is missing in the ros_planning version. The franka_ros package was quite a good inspiration ;-) I also had to adjust the `config/controllers.yaml` file to use the controller names from the franka_control configuration. 

To use the real gripper, I also changed the franka_ros `default_controllers.yaml` to include a JointTrajectoryController for the two finger joints. 

I also ran into some trouble with the duration scaling and margin settings in panda_moveit's `launch/trajectory_execution.xml` when I used the real robot/gripper in the lab. I increased the according values, but, honestly, I'm absolutely note sure if changing these variables until "it work's" is a clean solution...

Hope this supports your efforts in merging the two panda_moveit_configuration packages (#9) which I really appreciate.

fwalch, 2018-08-10:
@christgau Thanks for your detailed feedback!

The reason for the different planning groups is that we wanted to support two configurations, robot+gripper and just robot, from the same config (with the `load_gripper` arg on the launchfiles). In the ros-planning version, the gripper is part of the planning group and thus required.