moveit/panda_moveit_config #11: First release of panda_moveit_config from ros-planning

davetcoleman, 2018-11-09:
There have been updates to the panda_moveit_config since its merger from the original location https://github.com/frankaemika/franka_ros/

@rhaschke wants to make a release, but i want to make sure @fwalch and team is ok with this.

Also fyi @mlautman

fwalch, 2018-11-12:
> There have been updates to the panda_moveit_config since its merger

I can't see any updates since my PR was merged, but a release sounds good! I hope you won't run into problems with Kinetic (see also https://github.com/ros-planning/panda_moveit_config/pull/9#issuecomment-432936582), but otherwise let me know and we'll do a `franka_ros` release without `panda_moveit_config` and then you can release from here afterwards.

rhaschke, 2018-11-12:
I think, `panda_moveit_config` should be hosted at `franka_ros` and released by them.
See https://github.com/ros-planning/panda_moveit_config/pull/9#issuecomment-437818276.

rhaschke, 2018-11-13:
Release triggered: https://github.com/ros/rosdistro/pull/19460

fwalch, 2018-11-13:
For Kinetic, we had already released `panda_moveit_config` 0.6.0. This package therefore needs a higher version than that to be upgradable in Kinetic.

rhaschke, 2018-11-13:
Thanks for this hint!

rhaschke, 2018-11-13:
Release 0.7.0 triggered for Melodic (https://github.com/ros/rosdistro/pull/19461).

For Kinetic, @fwalch please remove/disable `panda_move_config` from `franka_ros` release.
`AssertionError - Duplicate package name 'panda_moveit_config' exists in repository 'panda_moveit_config' as well as in repository 'franka_ros'`.

rhaschke, 2018-11-14:
Release 0.7.0 for Kinetic triggered too: https://github.com/ros/rosdistro/pull/19472

closed 2018-11-14 by rhaschke