moveit/panda_moveit_config #13: Release into Lunar?

rhaschke, 2018-11-16:
Dear @davetcoleman,
Currently our Lunar Docker containers fail to build, because panda_moveit_config is not yet released into Lunar:
https://hub.docker.com/r/moveit/moveit/builds/bt27uxgqr446z6wwz6uywpn/

I think we should prepare a Lunar release as well. Do you agree?
However, first somebody should fix #12. Not sure what process generates the .srdf from .srdf.xacro...

davetcoleman, 2018-11-17:
Makes sense to me. @fwalch ?

rhaschke, 2018-11-19:
This needs a release of `franka_description` first.

fwalch, 2018-11-19:
We can easily release `franka_description`, but I'm not sure about the actual control parts (libfranka etc), as we don't use Lunar (and Yakkety/Zesty). Also Lunar will hit EOL in May already.

It could still make sense to release this for Lunar anyway, even without libfranka - you can use `demo.launch` and, with the new setup assistant, generate a URDF for Gazebo.

What do you think?

fwalch, 2018-11-19:
> Yes, you did the same for Melodic: Please only release franka_description if you feel more comfortable with this

Alright, I just wanted to clarify. For Melodic, we will release `libfranka` etc as well.

fwalch, 2018-11-19:
Opened https://github.com/ros/rosdistro/pull/19497

davetcoleman, 2018-11-21:
Docker images have been removed for Lunar fyi

closed 2019-04-05 by rhaschke