moveit/panda_moveit_config #17: MoveIt Assistant can't locate SRDF

aPonza, 2018-12-27:
I'm trying to debug some move_group related errors on my code and stumbled on this.

Running `rosrun moveit_setup_assistant moveit_setup_assistant --config_pkg ~/catkin_ws/src/panda_moveit_config/` or `roslaunch moveit_setup_assistant setup_assistant.launch` and choosing this package and `--inorder` results in error `Unable to locate the SRDF file: /home/<user>/catkin_ws/src/panda_moveit_config/config/panda.srdf`.

This seems to be due to file [.setup_assistant](https://github.com/ros-planning/panda_moveit_config/blob/3a1e58db644d3f61634bf9802c133f391b7be61b/.setup_assistant#L6) where the highlighted line should read something like `relative_path: config/panda_arm_hand.srdf.xacro`.

This seems to be due to the moving from the old panda_moveit_config in franka_ros, as I can see it was like [this](https://github.com/frankaemika/franka_ros/pull/26/commits/7f02b72e078654b28a799601101537c1e02e70fa#diff-b4df476127cbb048329ccfa8caa26d87L6) there.

Was this intended? This wasn't mentioned in the [PR](https://github.com/ros-planning/panda_moveit_config/pull/9).

byque, 2019-01-23:
I can confirm the same behaviour in my computer.

What I did was to create a new configuration using the set-up assistant and the panda_arm_hard.srdf.xacro found on the franka_ros repository which could be loaded without problems.

I must agree that moving panda_moveit_config results confusing in the current state.

aPonza, 2019-05-29:
This is fixed by #19 (it's `panda_arm` instead of `panda_arm_hand` but it doesn't seem to change much) but depends on #25.

rhaschke, 2019-06-02:
@aPonza, I guess you are using the Kinetic version, which is why you depend on #25?

closed 2019-07-16 by aPonza