moveit/panda_moveit_config #21: Unable to interact with moveit following tutorial

manestay, 2019-02-14:
I can't get step 2 to work following the guide [https://ros-planning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html](here) exactly as written. I have interact selected, but it just moves the camera, and no interactive markers appear. I only see the green and orange highlighted portions of the robot.

I see that the latest commit on melodic-devel was 4 days ago, whereas the tutorial was updated 22 days ago. My apologies if this is not the right place to ask.

manestay, 2019-02-14:
Posted in stack overflow since it's probably more relevant there [link](https://stackoverflow.com/questions/54696068/unable-to-interact-with-joints-following-official-moveit-tutorial)

closed 2019-02-14 by manestay

TZECHIN6, 2022-12-26:
Do you find a way to solve it? facing the same issue... seem everyone can interact with the robot once it started