moveit/panda_moveit_config #22: Changing end effector group breaks tutorial

naktamello, 2019-03-04:
on the kinetic-develop branch while running demo.launch,
I have noticed the interactive markers are no longer visible
after commit d97b987a7bfc4ec0d10fa01d1ae518a67daf31d4
What was the reason for changing the end effector parent group?

skasperski, 2019-03-04:
Running demo.launch on melodic does not work either, all robot links seem to be cluttered at the center point. It only shows the following warning:

`[ WARN] [1551693269.759478997]: Unable to update multi-DOF joint 'virtual_joint': Failure to lookup transform between 'world' and 'panda_link0' with TF exception: "world" passed to lookupTransform argument target_frame does not exist.`

krigi100, 2019-04-11:
Did you find a solution?
I have the same problem with another warning.
There are two (same) warnings:
```
[ WARN] [1554989812.752012929]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/rviz_ubuntu_9594_3039662777278291165/panda_arm/kinematics_solver_attempts' from your configuration.
```

The first one, can be deleted by removing "kinematics_solver_attempts: 5" in the "kinematics.yaml".
But I don't understand, why there is the second (same) warning. I can't find the parameter in the parameter list.

davetcoleman, 2019-04-17:
These warnings are very likely unrelated to the original question. However, regarding the second warning, did you restart roscore?

davetcoleman, 2019-04-17:
> TF exception: "world" passed to lookupTransform argument target_frame does not exist.

Sounds like you aren't publishing that transform with TF

mlautman, 2019-04-22:
@naktamello and @skasperski  I believe that PR's #23 and #24 should fix this issue. Are you using a source install of panda_moveit_config or a debian?

mlautman, 2019-04-22:
Both source and debian installs of panda_moveit_config's work fine for me running melodic on 18.04. Please re-open if you still have the issue after updating your debians.

closed 2019-04-22 by mlautman