moveit/panda_moveit_config #25: unify kinetic and melodic into master branch

rhaschke, 2019-04-05:
Kinetic and Melodic branches for this repo have diverged for no reason. Usually, there is no reason to distinguish ROS distros for a moveit_config package.

The latest kinetic release is missing 51b21a7356a3d11792f9fe4e5621837f618cca1f, which is the cause for issue https://github.com/ros-planning/moveit/issues/1426.

I suggest to continue development on a master branch only.

[image]

rhaschke, 2019-06-02:
@davetcoleman or @mlautman, could you please comment on this suggestion and then take the appropriate actions?

rhaschke, 2019-06-02:
In view of #31, which introduces an actual deviation between kinetic and melodic, I suggest
1. to _hard-reset_ `kinetic-devel` (which currently has two randomly chosen cherry-picks from melodic-devel) to `melodic-devel`
2. release those changes into both, kinetic and melodic (maybe also including #30)
3. and _subsequently_ branch off melodic-devel from kinetic-devel to handle #31.

mlautman, 2019-06-03:
Sounds like a good plan

rhaschke, 2019-06-04:
Done. Releases triggered:
* for Melodic: https://github.com/ros/rosdistro/pull/21451
(Not sure, why  @mlautman didn't release 0.7.2 when he created the release tag)
* for Kinetic: https://github.com/ros/rosdistro/pull/21450

closed 2019-06-04 by rhaschke