moveit/panda_moveit_config #28: Problem following tutorial: Franka entangled

fjulian, 2019-05-20:
Hi,
I am trying to follow the tutorial [here](https://ros-planning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html). Before starting, I successfully completed the [previous one](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html), which is a prerequisite.
Launching the demo in step one works fine. The Rviz window opens up as expected. However, after adding the motion planning plugin, the robot doesn't look like in the tutorial, but is entangled, as can be seen in the following image.
[image]
This screenshot was taken right after adding the plugin without changing any settings.

When I continue to follow the tutorial and arrive at step 3, I can interact with the robot. However, instead of being able to bring the robot into a more "normal" configuration, the joints just spin randomly, see this screenshot.
[image]

Could you tell me what I am doing wrong here? Any help is greatly appreciated!

I'm using Ubuntu 18.04 and installed ROS Melodic as described in the "Getting Started" tutorial.

Thanks in advance and kind regards.

gavanderhoorn, 2019-05-20:
Do you have a German-like locale configured on your machine? I assume so, as the screenshots show commas instead of dots in floating point numbers.

If so, this could be ros-planning/moveit#1050, which in the end was ros/urdfdom_headers#41.

That is -- afaik, but correct my if I'm wrong @simonschmeisser -- still not fixed.

fjulian, 2019-05-20:
Yes I had locale DE. I now changed to US and everything is working fine. Thank you!

closed 2019-05-20 by fjulian

gavanderhoorn, 2019-05-20:
Please vote for the bug report opened by @simonschmeisser linked in https://github.com/ros-planning/moveit/issues/1050#issuecomment-467436892.

Without more people complaining about this it's not going to get fixed and Melodic (and Bionic) will be with us for a long time ..