moveit/panda_moveit_config #33: Regression on the current kinetic-devel

aPonza, 2019-07-17:
**tl;dr** 6ae03fac5cf19f830f60e10a5bf669e0148a8e1f doesn't fix for me the virtual joint issue that was fixed for #22, and it's not the case of just propagating the publisher to another launch file.

### The issue

With the same exact process running, checking out panda_moveit_config 0.7.1 runs fine but the current kinetic-devel gives erratic behaviour and this diff in the output (I tried to keep it as short as possible:
```
--- /home/ap/Desktop/kinetic-devel
+++ /home/ap/Desktop/0.7.1
@@ -187,6 +187,7 @@
  * /position_joint_trajectory_controller/type: position_controll...
  * /robot_description: <?xml version="1....
  * /robot_description_kinematics/panda_arm/kinematics_solver: kdl_kinematics_pl...
+ * /robot_description_kinematics/panda_arm/kinematics_solver_attempts: 5
  * /robot_description_kinematics/panda_arm/kinematics_solver_search_resolution: 0.001
  * /robot_description_kinematics/panda_arm/kinematics_solver_timeout: 0.05
  * /robot_description_planning/joint_limits/panda_finger_joint1/has_acceleration_limits: False
@@ -229,6 +230,7 @@
  * /rosdistro: kinetic
  * /rosversion: 1.12.14
  * /rviz_ap_NUC7i7DNKE_15599_3838872138042140295/panda_arm/kinematics_solver: kdl_kinematics_pl...
+ * /rviz_ap_NUC7i7DNKE_15599_3838872138042140295/panda_arm/kinematics_solver_attempts: 5
  * /rviz_ap_NUC7i7DNKE_15599_3838872138042140295/panda_arm/kinematics_solver_search_resolution: 0.001
  * /rviz_ap_NUC7i7DNKE_15599_3838872138042140295/panda_arm/kinematics_solver_timeout: 0.05
 
@@ -288,13 +290,12 @@
 [ INFO] - [/move_group::PlanningSceneMonitor::startWorldGeometryMonitor::1025]: Starting world geometry monitor
 [ INFO] - [/move_group::PlanningSceneMonitor::startWorldGeometryMonitor::1041]: Listening to '/collision_object' using message notifier with target frame '/world '
 [ INFO] - [/move_group::PlanningSceneMonitor::startWorldGeometryMonitor::1056]: Listening to '/planning_scene_world' for planning scene world geometry
-[ INFO] - [/move_group::PointCloudOctomapUpdater::start::114]: Listening to '/camera/depth_registered/points' using message filter with target frame '/world '
+[ INFO] - [/move_group::PointCloudOctomapUpdater::start::114]: Listening to '/camera/depth_registered/points' using message filter with target frame '/panda_link0 '
 [INFO] - [/controller_spawner::main::192]: Controller Spawner: Loaded controllers: position_joint_trajectory_controller
 [ INFO] - [/move_group::PlanningSceneMonitor::startStateMonitor::1118]: Listening to '/attached_collision_object' for attached collision objects
 Context monitors started.
 [ INFO] - [/franka_control::FrankaHW::prepareSwitch::371]: FrankaHW: Prepared switching controllers to joint_position with parameters limit_rate=1, cutoff_frequency=100, internal_controller=joint_impedance
 [INFO] - [/controller_spawner::main::201]: Started controllers: position_joint_trajectory_controller
-[ERROR] - [/franka_control::main::206]: libfranka: Move command rejected: command not possible in the current mode!
 [ INFO] - [/move_group::OMPLInterface::OMPLInterface::55]: Initializing OMPL interface using ROS parameters
 [ INFO] - [/move_group::PlanningPipeline::configure::119]: Using planning interface 'OMPL'
 [ INFO] - [/move_group::FixWorkspaceBounds::FixWorkspaceBounds::53]: Param 'default_workspace_bounds' was not set. Using default value: 10
@@ -308,7 +309,6 @@
 [ INFO] - [/move_group::PlanningPipeline::configure::163]: Using planning request adapter 'Fix Start State Bounds'
 [ INFO] - [/move_group::PlanningPipeline::configure::163]: Using planning request adapter 'Fix Start State In Collision'
 [ INFO] - [/move_group::PlanningPipeline::configure::163]: Using planning request adapter 'Fix Start State Path Constraints'
-[ WARN] - [/move_group::CurrentStateMonitor::tfCallback::441]: Unable to update multi-DOF joint 'virtual_joint': TF has no common time between '/world' and 'panda_link0': 
 [ INFO] - [/franka_control::<lambda::131]: Recovered from error
 [ INFO] - [/move_group::MoveItSimpleControllerManager::MoveItSimpleControllerManager::145]: Added FollowJointTrajectory controller for position_joint_trajectory_controller
 [ INFO] - [/rviz_ap_NUC7i7DNKE_15599_3838872138042140295::RenderWindow* rviz::RenderSystem::makeRenderWindow::441]: Stereo is NOT SUPPORTED
@@ -352,7 +352,6 @@
 
 [ INFO] - [/process_manager::core::RobotModel::buildModel::93]: Loading robot model 'panda'...
 [ INFO] - [/process_manager::core::RobotModel::buildModel::93]: Loading robot model 'panda'...
-[ WARN] - [/move_group::PlanningSceneMonitor::updateSceneWithCurrentState::1255]: The complete state of the robot is not yet known.  Missing virtual_joint
 [ INFO] - [/process_manager::planning_interface::MoveGroupInterface::MoveGroupInterfaceImpl::MoveGroupInterfaceImpl::178]: Ready to take commands for planning group panda_arm.
 [ INFO] - [/process_manager::Panda::Panda::403]: Waiting for a subscriber on topic /world_to_panda_link0.
 [ INFO] - [/static_transform_aggregator::StaticTransformAggregator::sendAllTransforms_::98]: Currently static broadcasting:
@@ -376,12 +375,12 @@
 [ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::185]: RRTConnect: Starting planning with 1 states already in datastructure
 [ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::185]: RRTConnect: Starting planning with 1 states already in datastructure
 [ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::185]: RRTConnect: Starting planning with 1 states already in datastructure
-[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::292]: RRTConnect: Created 5 states (2 start + 3 goal)
-[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::292]: RRTConnect: Created 4 states (2 start + 2 goal)
-[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::292]: RRTConnect: Created 5 states (2 start + 3 goal)
-[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::292]: RRTConnect: Created 5 states (2 start + 3 goal)
-[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::129]: ParallelPlan::solve(): Solution found by one or more threads in 0.013718 seconds
-[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::185]: SimpleSetup: Path simplification took 0.005437 seconds and changed from 3 to 2 states
+[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::292]: RRTConnect: Created 4 states (2 start + 2 goal)
+[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::292]: RRTConnect: Created 4 states (2 start + 2 goal)
+[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::292]: RRTConnect: Created 4 states (2 start + 2 goal)
+[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::292]: RRTConnect: Created 4 states (2 start + 2 goal)
+[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::129]: ParallelPlan::solve(): Solution found by one or more threads in 0.012583 seconds
[... 58 lines cut ...]
-[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::292]: RRTConnect: Created 6 states (2 start + 4 goal)
-[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::292]: RRTConnect: Created 5 states (2 start + 3 goal)
-[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::292]: RRTConnect: Created 5 states (3 start + 2 goal)
-[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::129]: ParallelPlan::solve(): Solution found by one or more threads in 0.063474 seconds
-[ INFO] - [/move_group::OMPLPlannerManager::OMPLPlannerManager::185]: SimpleSetup: Path simplification took 0.002666 seconds and changed from 3 to 2 states
-[ INFO] - [/move_group::FixStartStatePathConstraints::adaptAndPlan::92]: Planned to path constraints. Resuming original planning request.
 [ INFO] - [/move_group::core::RobotModel::buildModel::93]: Loading robot model 'panda'...
-[ WARN] - [/move_group::JointConstraint::configure::179]: Joint panda_joint2 is constrained to be above the maximum bounds. Assuming maximum bounds instead.
-[ WARN] - [/move_group::JointConstraint::configure::179]: Joint panda_joint4 is constrained to be above the maximum bounds. Assuming maximum bounds instead.
-[ WARN] - [/move_group::JointConstraint::configure::179]: Joint panda_joint2 is constrained to be above the maximum bounds. Assuming maximum bounds instead.
-[ WARN] - [/move_group::JointConstraint::configure::179]: Joint panda_joint4 is constrained to be above the maximum bounds. Assuming maximum bounds instead.
[... cut ...]
- addition of virtual joint: I think this might be the cause, as everything seems to stem from `Unable to update multi-DOF joint 'virtual_joint': TF has no common time between '/world' and 'panda_link0'` (which is incidentally different from [this](https://github.com/ros-planning/panda_moveit_config/issues/22#issuecomment-469191161)).

Seems to be this [this problem](https://github.com/ros-planning/moveit/issues/1399#issuecomment-473209570) but I tried to apply the fix to my own higher level launch file or directly to panda_control_moveit_rviz, and it doesn't solve the issue, whereas like the comment says if I change `floating` to `fixed` it does work.

rhaschke, 2019-08-30:
@aPonza, could you check, whether https://github.com/ros-planning/moveit_tutorials/pull/387 fixes this issue? If so, please close this issue.

aPonza, 2019-09-09:
It doesn't seem to, the output is mainly the same. Just to double check, I had to clone moveit, moveit_resources and moveit_tutorials, stash my changes in panda_moveit_config (which wasn't modified) and rebuild the workspace to get this fix in, correct? I don't understand how a fix in a tutorial would affect the issue I'm seeing with the floating/fixed joint to be honest, but I'll have time to look a bit more into it in some days' time hopefully.

rhaschke, 2019-09-09:
https://github.com/ros-planning/moveit_tutorials/pull/387 addressed an issue in the `motion_planning_api_tutorial`, which directly uses the C++ API of MoveIt, circumventing the ROS interface provided via the `move_group` node.

Please provide exact commands to reproduce your issue. You didn't even mention, which of the tutorials you tried.

aPonza, 2019-09-09:
I'm not trying tutorials, I'm using my own nodes and launch file of a pick and place process.  It's exactly the same roslaunch command with floating vs fixed. I'll make it reproducible, you're right. It'll take some time, I'm working on another thing at the moment.

rickstaa, 2021-11-02:
@rhaschke Maybe we can close this issue since kinetic is EOL?

rhaschke, 2021-11-02:
The virtual joint should be type `fixed`. This doesn't require a virtual joint broadcaster, which is not provided by the real robot.

closed 2021-11-02 by rhaschke