moveit/panda_moveit_config #37: Remapping of joint_states intended?

GunnarMaer, 2019-08-02:
Hello!

I stumbled on the line which remaps the joint states:
[`<remap from="/joint_states" to="/joint_states_desired" />`](https://github.com/ros-planning/panda_moveit_config/blob/e3a431a9ca1421d254e7d121ece6620d4a9254f1/launch/move_group.launch#L78)
(l. 78 in move_group.launch)

Is this line intended? I am working with a real panda robot and was creating launch files for two different usecases. In both cases, I removed this line, as it seemed to be not only not-necessary; but also hindering, as it prevented move_group from getting the real joint states. Is this desired?  

It looks like I'm not the only one who came across this ([see here](https://answers.ros.org/question/302674/didnt-received-robot-state-joint-angles-with-recent-timestamp-within-1-seconds-check-clock-synchronization-if-your-are-running-ros-across-multiple/)), so it would be great if you could look over it and check which way it should be.

Thank you!

fwalch, 2019-08-02:
Some explanation for why this is required is provided in https://github.com/ros-planning/panda_moveit_config/pull/9#issuecomment-412055080.
I'll try to expand on this, but in the meantime, maybe the [docu](https://frankaemika.github.io/docs/libfranka.html#under-the-hood) can help to explain why there is a difference between measured joint states `q` and desired joint states `q_d`.

GunnarMaer, 2019-08-12:
Alright. Thanks for the clarification. Good to know that this is intended. 
To be honest, this is rather confusing. My issues were probably related to mixing up the joint state publishers, but I will go through my launch files once more.

Since I'm using franka_ros, I just skimmed through the part of the docu on libfranka.

closed 2019-08-12 by GunnarMaer