moveit/panda_moveit_config #49: missing robot state for virtual joint panda_joint8 - Trying to control panda from RVIZ

FirefoxMetzger, 2020-01-14:
I'm running ROS Kinetic and am connected to a physical panda arm. As far as I can tell, that setup works fine, as I can control the robot from the Franka Emika GUI just fine.

I'm trying to run the launch file `panda_control_moveit_rviz.launch` which opens RVIZ and shows a virtual representation of the robot in it's current (physical pose). When I guide the physical robot, the pose is also updated correctly within RVIZ. I am building both `franka_ros` and `panda_moveit_config` from the current master branch.

I am not seeing the interactive markers (see https://github.com/ros-planning/moveit/issues/1399) and I am seeing a warning: `The complete state of the robot is not yet known. Missing virtual_joint, panda_joint8`. I know that joint8 doesn't exist on the physical robot - likely it has been introduced for the sake of the tutorials (?).

Any ideas regarding a fix?

rhaschke, 2020-01-14:
`joint8` is a fixed joint that exists in the official URDF:
https://github.com/frankaemika/franka_ros/blob/f7f00b6d9678e59e6f34ccc8d7aad6491b42ac80/franka_description/robots/panda_arm.xacro#L164-L169

The `virtual_joint` is indeed added by MoveIt's SRDF to place the robot in the world. To define it,
you should use a `static_transform_publisher`:
https://github.com/ros-planning/panda_moveit_config/blob/4a16a4fcdbf5cc7975937faa1b0ebcf8d41b365d/launch/demo.launch#L29

If this solves your issue, please file a corresponding PR to fix the launch file.

FirefoxMetzger, 2020-01-14:
> The virtual_joint is indeed added by MoveIt's SRDF to place the robot in the world. To define it,
you should use a static_transform_publisher:

I will try this after lunch. I found a more hacky solution, which was to change the virtual joint from `floating` to `fixed`, although I am not familiar enough with the code base yet and don't know if that breaks anything else. Interestingly, introducing the hack resolved the warning for both joints.

rhaschke, 2020-01-14:
I think it is a bug that `joint8` was mentioned in the error message...

FirefoxMetzger, 2020-01-14:
@rhaschke Done. Your suggestion resolves the warning for both joints as well.

I created two PRs for melodic and kinetic respectively.

PR #51 (2020-01-14): added static transform publisher
Adds a static publisher to the launch file to resolve a warning about missing joints and bring back the interactive markers in RVIZ.

resolves #49
closed 2020-01-14 by rhaschke