moveit/panda_moveit_config #71: No collision geometry

JhuaDuan, 2020-09-23:
When i was ran :` roslaunch panda_moveit_config panda_control_moveit_rviz.launch load_gripper:=true robot_ip:=172.16.0.2`. Some warn occurred as following:

> [ WARN] [1600866644.391617690]: Link panda_leftfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
> [ WARN] [1600866644.392008313]: Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.

v4hn, 2020-09-23:
This is a bug in the current upstream robot model [that was pointed out multiple times already](https://github.com/ros-planning/panda_moveit_config/pull/35#issuecomment-671279013). They did not fix it yet though and you should file an issue [there](https://github.com/frankaemika/franka_ros) instead.

That being said, the warning can be safely ignored unless you rely on these specific collision bodies, because other links have collision geometry that encloses the empty ones.

closed 2020-09-23 by v4hn

JhuaDuan, 2020-09-23:
@v4hn Thanks for your help. This bug causes the gripper to disconnect,  since I can't ignore this.