moveit/panda_moveit_config #72: # Noetic Release (0.8.0)

tylerjw, 2020-09-30:
# Noetic Release (0.8.0)

`franka_description` was released: https://index.ros.org/p/franka_description/#noetic

Before we cut the release we should regenerate this based on the master version of the moveit setup assistant?  Is there any documentation for the process of generating this package or is it done using all default settings in the setup assistant?  @rhaschke 

### Checklist
- [ ] Update based on master branch of moveit setup assistant
- [ ] Make `noetic-devel` branch
- [ ] Run ROS buildfarm prerelease test
- [ ] Update changelogs
- [ ] Bloom
- [ ] Write discourse post

tylerjw, 2020-09-30:
Before anyone points it out, I found that the tutorial for setup assistant describes how to create this package.  I'll do that now and create the PR for it.

YijiongLin, 2021-01-12:
Hi gentleman,

Currently, I'm working on the Franka panda arm (both the real and the simulated one), and I'm studying the possibility of using that arm with the franka_ros pkg. I tend to work on ROS Noetic since it would be great if I could work on these code with Python 3 instead of Python 2.

I was redirected to [here](https://github.com/ros-planning/panda_moveit_config/issues/72) from [issue #2341](https://github.com/ros-planning/moveit/issues/2341), in which it said panda_moveit_config (0.8.0) is a future plan (not build yet). And [https://github.com/frankaemika/franka_ros/issues/119](url) also said the _ranka_ros 0.7.1 for Noetic_ will be fully released once [issue 72 (panda_moveit_config (0.8.0) ) ](https://github.com/ros-planning/panda_moveit_config/issues/72) is done. However I came across [this website](https://index.ros.org/p/franka_ros/#noetic) in which it seems  _ranka_ros 0.7.1 for Noetic_ has been released (I'm really not sure if things being put on this website means that have been released).

To be short, I know [Moveit Noetic](url) is released, but still have several minor questions that might need your help:
1. Is franks_ros Noetic 0.7.1 released? If yes, which website gives more details on it?
2.If I under things correctly, panda_moveit_config should refer to a folder that includes all the files decribing the panda arm, and this config folder can be generated by the Moveit Settup Assistant if we have the panda arm urdf files. So why would it still a problem? (I'm a newbie to Moveit, please forgive me if these problems bother you)

Tobias-Fischer, 2021-01-27:
Hiya,
It would be great to see this package released for noetic. Is there anything I can help to make this happen?

tylerjw, 2021-01-27:
Thank you for asking, this PR: https://github.com/ros-planning/panda_moveit_config/pull/74 is where we are stuck.  I started the work of using the latest panda config from franka for this package (and therefore for the moveit tutorials) however I've had my time eaten up by other work and have not had time to finish it.  A couple of other people have helped on that PR and you are welcome to test with it and try helping with some of the issues laid out there.  If we can get that PR finished we can then release `panda_moveit_config` for noetic.

> panda_moveit_config should be referred to as a folder that includes all the files describing the panda arm, and this config folder can be generated by the Moveit Setup Assistant if we have the panda arm URDF file

The primary problem is it is not that simple sadly.  There are two things that make this more complex.  

The first is that the fraka description was updated to have the collision geometry that the robot uses internally in its controller to prevent self collisions.  This was probably done because people were using actual panda robots with this config and were frustrated when the frank hardware controller would not perform trajectories that were determined to be collision-free by moveit and could be performed in simulation.  With the updated description moveit will not generate paths that can't be executed on real robots which is good.  What is bad is that this collision mesh is really course and prevents many movements that a panda could otherwise perform.  Some of those paths (and poses) are/were part of the tutorials and default poses we had before.  Much of those issues are solved in that PR by changing default poses.  There may still be more effects of this we don't realize yet.

The second issue is that creating this package, `panda_moveit_config`, is not as simple as just using moveit_setup_assistant.  There are many manual changes that have been applied to this package over time to facilitate various tutorials in the moveit tutorials and other use-cases people have for this package.  Sadly this is where most of the unfinished work is.

In order to finish that PR someone needs to get all the tutorials working and understand the current state of the `panda_moveit_config` package to forward features into the newly generated one manually.  I still don't understand some of the stuff around xacro files and similar issues.

If you would like we could really use help getting that PR finished and then I'll be able to release this package for Noetic.

--

As to what I've been working on.  Much of that is ros2, and therefore moveit2.  If it is possible for you to do your project with ros2, we have more internal funding and more time to support moveit2 now and in the future.  What that means is that fixing issues like this is a higher priority (and therefore will be easier for us to get done in a timely way) for us when they affect moveit2.  The future of ROS is ROS2.  I will continue to support Noetic and releases for it as a best-effort but will have very limited time to do so sadly.

Tobias-Fischer, 2021-01-27:
Ahh gosh, that is much more involved than what I had imagined. I am not very familiar with moveit in general, so I think this task would eat up too much of my time - sorry. The main reason why I am keen to get this package to work is to get franka_ros into noetic, which we in turn would like to build within our RoboStack project, which combines ROS with conda: https://github.com/RoboStack/ros-noetic