moveit/panda_moveit_config #76: run-time error with franka_ros example

christoph-jaehne, 2021-03-15:
With the current apt version of ros-melodic-panda-moveit-config and the current franka_ros, when launching e.g. the move to start example: 

```bash
roslaunch franka_example_controllers move_to_start.launch
```
I get the following error message:

```bash
[ERROR] [1615826543.912941470]: Exception while loading planning adapter plugin 'default_planner_request_adapters/ResolveConstraintFrames': According to the loaded plugin descriptions the class default_planner_request_adapters/ResolveConstraintFrames with base class type planning_request_adapter::PlanningRequestAdapter does not exist. Declared types are  default_planner_request_adapters/AddIterativeSplineParameterization default_planner_request_adapters/AddTimeOptimalParameterization default_planner_request_adapters/AddTimeParameterization default_planner_request_adapters/Empty default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints default_planner_request_adapters/FixWorkspaceBounds
```

My installed packages are all up to date...

christoph-jaehne, 2021-03-15:
The type `ResolveConstraintFrames` seems to be missing in the current set of apt packages...

rhaschke, 2021-03-15:
Consider this as a warning. The MoveIt capability `ResolveConstraintFrames` is only available in Noetic / MoveIt master, but the panda_moveit_config package is used across Melodic and Noetic.

felixvd, 2021-03-16:
So in summary, you can ignore the warning. 

The `ResolveConstraintFrames` is needed for the Subframes feature, which is not available on Melodic.

fwalch, 2021-03-18:
Thanks for your help! As you said, this only has the effect of a warning and does not represent an actual error, which we did not realize at first. The actual issue was that #35 has not been released yet (see #75), so this can be closed.

closed 2021-03-18 by christoph-jaehne