moveit/panda_moveit_config #77: Release 0.7.x for Noetic

fwalch, 2021-03-18:
As #72 seems to require additional work, would it be possible to release the existing `melodic-devel` branch for Noetic? I only performed basic tests, but it seems to work without issues on Noetic. Our `franka_example_controllers` package depends on `panda_moveit_config`, so we cannot publish it to the ROS repos unless `panda_moveit_config` is in there as well.

tylerjw, 2021-03-18:
I assigned this to myself and will work on getting this done.

tylerjw, 2021-03-25:
https://github.com/ros/rosdistro/pull/28831

fwalch, 2021-03-25:
Thank you! However, the 0.7.4 tag is outdated compared to `melodic-devel`, and I think there will be problems with the `franka_ros` version already in the Noetic repos. I was hoping you could release the current `melodic-devel` branch. Sorry for not being clear.

Related for Melodic: #75

tylerjw, 2021-03-25:
Oh, I'm sorry I misunderstood.  I can cut a release of the latest for noetic.

tylerjw, 2021-03-25:
https://github.com/ros/rosdistro/pull/28851

tylerjw, 2021-03-26:
It was merged into bloom.  This will go out in the next sync.  I hope that fixes your issue.

fwalch, 2021-03-31:
It does, thank you!

closed 2021-03-31 by fwalch

rickstaa, 2021-04-08:
@tylerjw Thanks a lot for this release!