moveit/panda_moveit_config #89: Noetic release

rickstaa, 2021-10-21:
Me and @rhaschke just released the initial version of the [Noetic branch](https://github.com/ros-planning/panda_moveit_config/tree/noetic-devel). This issue will be used for the rest of the release process.

## TODOS

- [ ] As pointed out by @tahsinkose in https://github.com/ros-planning/panda_moveit_config/commit/c46c81c8a1e0da1288d8ef4462b8daa10df37213#r58426734 the version of the [franka_gazebo](https://github.com/frankaemika/franka_ros/tree/develop/franka_gazebo) package should be pinnned to v0.8.1 or higher.
- [ ] All the tutorials need to be re-tested.

## Upstream pull requests that are required

The following upstream pull requests need to be merged for the Noetic branch to be warning and error free:

- [ ] frankaemika/franka_ros#151 needs to be merged to get rid of the `Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.` warning. 
- [ ] frankaemika/franka_ros#173 needs to be merged before MoveIt can correctly control the gripper. Currently, MoveIt can only open and close the gripper but not set a desired width.
- [ ]  frankaemika/franka_ros#154 could improve the self-collision behaviour. When this pull request is merged, we can 'use_collision_mesh' arg to let MoveIt plan with the more accurate collision meshes. In that case aa24fae is then not needed anymore. If we however decide to do this we have to keep in mind that MoveIt can plan paths that are not possible on the real robot.
- [ ] The Noetic branch has been created using the `moveit_setup_assistant` found in [tree/82378877879f7d40727819b1c3345eeb4308d98b](https://github.com/ros-planning/moveit/tree/82378877879f7d40727819b1c3345eeb4308d98b). The current MoveIt debian package does not include the `pilz_industrial_motion_planner`. As a result, MoveIt first has to create a new release before it can be used with the debian packages.

The following upstream pull request is required for the simulation to work:

- [ ] frankaemika/franka_ros#181 needs to be merged for us to use the `position_joint_trajectory_controller`.

## Moveit_tutorial changes

I found the following changes that need to be performed to the [moveit_tutorials](https://github.com/ros-planning/moveit_tutorials):

- [ ]  [setup_assistant_tutorial.html](https://ros-planning.github.io/moveit_tutorials/doc/setup_assistant/setup_assistant_tutorial.html):
    -  [ ] Should have updated pictures (i.e., menu order was changed).
    -  [ ] Panda description file was changed to be `/franka_description/robots/panda.urdf.xacro` the hand is now enabled by using the `hand:=true` xacro parameter.
    -  [ ] I think the documentation should mention the `panda_finger_joint2` as the passive joint.
    -  [ ] The virtual joint doesn't appear to be needed in the panda_arm group. Maybe remove it as it will cause errors when using a gazebo simulation?
-  [ ] Maybe the [panda_control_moveit_rviz.launch] file should be documented in the documentation.
-  [ ] The `demo_chomp.launch` file was added to the [panda_moveit_config](https://github.com/ros-planning/panda_moveit_config/) repository, but this file is not used in the moveit_tutorials. 
-  [ ] The lerp planner documentation talks about a `lerp_example.launch` file but this file is not included I think this should be changed to refer the `demo_lerp.launch` file.

## Moveit_setup_assistant changes

I found several improvements that can be made to the MSA.

-   [ ] Maybe we can add the ability to use action services as controllers (i.e. https://github.com/rickstaa/panda_moveit_config/blob/47073b308a66a322146c7544f943533575f99d9e/config/panda_gripper_moveit_controllers.yaml#L14-L20).
-   [ ] Maybe we should move `controllers_list` from `ros_controllers.yaml` into its own file as these are only needed for MoveIt while the settings in the `ros_controllers.yaml` file should only contain settings related to the ROS control (see https://github.com/rickstaa/panda_moveit_config/pull/45 for a example).
-   [ ] The code that loads the `robot_description` in the [gazebo.launch` template](https://github.com/ros-planning/moveit/blob/a0ee2020c4a40d03a48044d71753ed23853a665d/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/gazebo.launch#L15) from:
    ```yaml 
    <param name="robot_description" textfile="$(arg urdf_path)" />
    ````
    to
    ```yaml
    <param name="robot_description" command="$(find xacro)/xacro $(arg urdf_path) hand:=$(arg load_gripper) gazebo:=true"/>
    ```
    if a `xacro` file is supplied by the user. 
- [ ] As the https://github.com/frankaemika/franka_ros was changed, the code that creates the gazebo simulation should be adjusted. The https://github.com/frankaemika/franka_ros repo now includes all the components to create a gazebo simulation, but loads the robot onto the `/panda` namespace. As a result, the current templates don't work anymore. We should apply the changes found in https://github.com/rickstaa/panda_moveit_config/pull/40 to the templates in the MSA.  
- [ ] The STOMP templates should be added to the MSA.
- [ ] The `stomp_demo` and `chomp_demo` files are not created by the MSA. Maybe we should add these files?
- [ ] The `ompl-chomp_planning_pipeline.launch.xml` should be added to the MSA (see https://ros-planning.github.io/moveit_tutorials/doc/chomp_planner/chomp_planner_tutorial.html#using-chomp-as-a-post-processor-for-ompl).
- [ ] The CHOMP algorithm contains several commented out parameters that might be removed (see https://github.com/ros-planning/moveit/blob/d57721cf8503fe893347196cc1e1f9249ec57777/moveit_planners/chomp/chomp_motion_planner/include/chomp_motion_planner/chomp_parameters.h#L81-L88). 
- [ ] Maybe we also want to add the remaping found in https://github.com/rickstaa/panda_moveit_config/pull/54 into the `moveit_setup_assistant` template.
- [ ] The CHOMP `ridge_factor` parameter should be changed to `0.0` since this is the default.
- [ ] The CHOMP `enable_failure_recovery` parameter should be changed to `false` since this is the default.

## Other upstream problems that still exist

- [ ] I did not test the workings of the TrajOpt planner since it did not build on my system (see https://github.com/ros-planning/moveit/pull/1626#issuecomment-933393754).
- [ ] The STOMP planner seems to fail on ROS Noetic due to a [upstream bug](https://github.com/ros-industrial/industrial_moveit/issues/53).

## Migration guide

For future reference, the following steps were performed to release the Noetic branch:

1. I created the initial panda robot configuration using the [moveit_setup_assistant](https://github.com/ros-planning/moveit/tree/master/moveit_setup_assistant) (see 8566b70f5e464e6fcb1c7d5664ff6e6e5202876e). In this branch, [tree/82378877879f7d40727819b1c3345eeb4308d98b](https://github.com/ros-planning/moveit/tree/82378877879f7d40727819b1c3345eeb4308d98b) was used. Similar to https://github.com/ros-planning/panda_moveit_config/pull/74 I changed some posed to prevent self collisions.

2. I modified the `panda.srdf` file that is created by the `moveit_setup_assistant` such that we can programmatically enable the gripper (see f0c2bca91d14a5b4cd7192c189d253f4d1fb6407). This needs to be done, since the `moveit_setup_assistant` currently doesn't propagate `xacro` arguments you supply to the `urdf.xacro` file (see [this issue](https://github.com/ros-planning/moveit/issues/1691)). We therefore need to manually add a way to enable or disable the gripper. This can be done by creating a `panda.srdf.xacro` file.

[... 19 lines cut ...]

### Files that were manually created

#### Config folder

-   panda.srdf.xacro
-   hand.xacro
-   panda_arm.xacro
-   lerp_planning.yaml
-   stomp_planning.yaml
-   trajopt_planning.yaml
-   panda_controllers.yaml
-   panda_gripper_controllers.yaml

#### launch

-   demo_stomp.launch (Was not present in the melodic-devel branch).
-   demo_chomp.launch (Not used in the moveit_tutorials).
-   demo_lerp.launch (Used in the moveit_tutorials).
-   lerp_planning_pipeline.launch.xml
-   ompl-chomp_planning_pipeline.launch.xml
-   panda_gripper_moveit_controller_manager.launch.xml
-   run_benchmark_trajopt.launch
-   stomp_planning_pipeline.launch.xml
-   trajopt_planning_pipeline.launch.xml

rhaschke, 2021-11-05:
@v4hn, the results of our phone-conf meeting with Franka are summarized [here](https://github.com/rhaschke/panda_moveit_config/issues/12).
@rickstaa, I addressed various issues of your list in https://github.com/ros-planning/moveit/pull/2932, https://github.com/ros-planning/moveit/pull/2945, https://github.com/ros-planning/moveit/pull/2946.
I'm tempted to recreate the noetic-devel branch starting from these latest MSA templates.

rickstaa, 2021-11-12:
Closed because this has been replaced by #96.

>The Gazebo-capable URDF generated by MSA doesn't show the robot in Gazebo, when using moveit_resources_panda_moveit_config. However, it seems to be there. It's just not visible?!

Has been addressed in https://github.com/rickstaa/moveit/tree/add_MSA_write_gazebo_urdf_feature. The MSA now checks whether the specified `urdf` is Gazebo compatible and otherwise saves it into the generated `<ROBOT>_moveit_config`. I will perform some last tests Sunday and add the last warning popup. This popup will come up when a supplied `urdf` is invalid and lets people know about the file generation behaviour and points to http://gazebosim.org/tutorials?tut=inertia .

@rhaschke Can you double-check to see if there are still todos missing from #69. I will also address https://github.com/ros-planning/moveit/pull/2955 on Sun.

closed 2021-11-12 by rickstaa

rhaschke, 2021-11-12:
> Has been addressed in https://github.com/rickstaa/moveit/tree/add_MSA_write_gazebo_urdf_feature. The MSA now checks whether the specified `urdf` is Gazebo compatible and otherwise saves it into the [...]

I think it is reasonable to provide an option to save the generated, Gazebo-compatible URDF. However, I suggest providing a save button in the corresponding widget, as the .urdf file will be saved in an external (non moveit-config) package. All other files listed in the config-files widget, are related to the moveit-config package only.

However, my problem wasn't that I couldn't save the file. Of course, I did that. 
Nevertheless, Gazebo doesn't visualize the robot for me (although it is loaded and simulated). I guess, some material properties are not correctly configured for Gazebo:
[image]

rickstaa, 2021-11-12:
Strange, I remember that the urdf worked for me when using the steps in the tutorials [moveit_setup_assistant](https://ros-planning.github.io/moveit_tutorials/doc/setup_assistant/setup_assistant_tutorial.html#step-9-gazebo-simulation). I will investigate this weekend.

The current implementation is as follows:
- If the simulation widget is not used, no `gazebo_panda.urdf` will be added to the moveit_config folder. The file will be shown but is disabled. The demo launch file is added but uses the original urdf. My plan, however, was only to add the gazebo launch files when the simulation widget is used. 
- If the simulation widget is used and the provide `urdf` is Gazebo compatible, no `gazebo_panda.urdf` will be added to the moveit_config folder. The file will also be disabled in the file overview.
- if the widget is used by provided `urdf` is not Gazebo compatible, the new gazebo urdf is added to the moveit_config, and the gazebo launch files will use this urdf while the other launch files use the original urdf. My plan was here to show a warning.

I can, however, change it in any way we like. I will create a draft pull request so we can discuss it there. My main goal was to fix the gazebo demo launch files by providing a have to automatically add the gazebo urdf when the provided urdf is not Gazebo compatible.

rhaschke, 2021-11-12:
As I said, I generally like and appreciate your proposal. We can use the generated gazebo-compatible file in two ways:
- Overwrite the original file in the robot's description package. This option should be offered in the simulation widget.
- Alternatively, create the corresponding file as part of the moveit_config package. In that case, it should be listed in the config file widget. 
 
demo_gazebo.launch should use one or the other file. Both options are mutually exclusive.

rickstaa, 2021-11-16:
@rhaschke I took your suggestions into account. You can find the tested version of my proposal at https://github.com/ros-planning/moveit/pull/2960.