moveit/panda_moveit_config #96: Noetic release

rhaschke, 2021-11-08:
I reworked (and force-pushed) the `noetic-devel` branch once more as announced in https://github.com/ros-planning/panda_moveit_config/issues/89#issuecomment-961815149.

The new branch can be considered a release candidate, but relies on several open PRs in franka_ros (also see https://github.com/frankaemika/franka_ros/issues/193) and MoveIt, namely:
- [ ] https://github.com/ros-planning/moveit/pull/2938
  allowing collision checking to be disabled by default for specific links
- [ ] https://github.com/ros-planning/srdfdom/pull/97
  extending the syntax of SRDF to specify these defaults (and exceptions)
- [ ] https://github.com/frankaemika/franka_ros/pull/189
  defining detailed (`panda_hand`) and coarse (`panda_hand_coarse`) collision models for environmental and self collisions resp.
- [ ] https://github.com/frankaemika/franka_ros/pull/183
- [ ] https://github.com/frankaemika/franka_ros/pull/187
- [ ] https://github.com/frankaemika/franka_ros/pull/190
- [ ] https://github.com/frankaemika/franka_ros/pull/186
- [ ] https://github.com/frankaemika/franka_ros/pull/194
- [ ] https://github.com/frankaemika/franka_ros/pull/173

Other open TODOs (@rickstaa, can you work on these?):
- [ ] Include configs for other planners (CHOMP, STOMP, LERP, TrajOpt) into MoveIt Setup Assistant. This requires:
  - Adding files `launch` and `config` folders in [`MSA templates`](https://github.com/ros-planning/moveit/blob/7f30892cbcc95cb5fec1be76edec9d8846b1d6b8/moveit_setup_assistant/templates/moveit_config_pkg_template)
  - Registering them in corresponding [`launch`](https://github.com/ros-planning/moveit/blob/25a63b920adf46f0a747aad92ada70d8afedb3ec/moveit_setup_assistant/src/widgets/configuration_files_widget.cpp#L368-L370) and [`config`](https://github.com/ros-planning/moveit/blob/25a63b920adf46f0a747aad92ada70d8afedb3ec/moveit_setup_assistant/src/widgets/configuration_files_widget.cpp#L242-L244) sections of [MSA's file-config widget](https://github.com/ros-planning/moveit/blob/25a63b920adf46f0a747aad92ada70d8afedb3ec/moveit_setup_assistant/src/widgets/configuration_files_widget.cpp) - typically using the [`copyTemplate`](https://github.com/ros-planning/moveit/blob/25a63b920adf46f0a747aad92ada70d8afedb3ec/moveit_setup_assistant/src/widgets/configuration_files_widget.cpp#L422) method.
- [ ] Verify and update `moveit_tutorials`

rickstaa, 2021-11-10:
@rhaschke I am working through the latest todos in #89 so that we can replace it with #96. I will add the tests and changes we have to do for the tutorials below:

## Moveit_tutorials todos

@tylerjw already performed some tests for his solution with the [moveit_tutorials](https://github.com/ros-planning/panda_moveit_config/pull/74#issuecomment-703139739). We however need to re-perform these tests for the new Noetic release. Please first check https://github.com/ros-planning/panda_moveit_config/pull/74#issuecomment-703139739 before testing one of the following tutorials:

- [x] Getting Started
    - [ ] Works but two small pull requests are requried when the noetic-branch is ready: https://github.com/ros-planning/moveit_tutorials/pull/673 and https://github.com/ros-planning/moveit/pull/2924.
- [ ] MoveIt Quickstart in RViz
- [ ] Move Group C++ Interface
- [ ] Move Group Python Interface
- [ ] MoveIt Commander Scripting
- [ ] Robot Model and Robot State
- [ ] Planning Scene
- [ ] Planning Scene Monitor
- [ ] Planning Scene ROS API
- [ ] Motion Planning API
- [ ] Motion Planning Pipeline
- [ ] Creating MoveIt Plugins
- [ ] Visualizing Collisions
- [ ] Time Parameterization
- [ ] Planning with approximated Constraint Manifolds
- [ ] Pick and Place
- [ ] MoveIt Grasps
- [ ] MoveIt Task Constructor
- [ ] MoveIt Deep Grasps
- [ ] Subframes
- [ ] MoveItCpp Tutorial
- [ ] Using Bullet for Collision Checking
- [ ] Mesh Filter with UR5 and Kinect
- [ ] MoveIt Setup Assistant
- [ ] URDF and SRDF
- [ ] Low Level Controllers
- [ ] Preception Pipeline Tutorial
- [ ] IKFast Kinematics Solver
- [ ] TRACK-IK Kinematics Solver
- [ ] OPW Kinematics Solver for Industrial Manipulators
- [ ] Kinematics Configuration
- [ ] Custom Constraint Samplers
- [ ] Representation and Evaluation of Constraints
- [ ] OMPL Planner
- [ ] CHOMP Planner
- [ ] STOMP Planner
- [ ] TrajOpt Planner
- [ ] Pilz Industrial Motion Planner
- [ ] Sequence of multiple segments
- [ ] Planning Adapter Tutorials
- [ ] Warehouse - Persistent Scenes and States
- [ ] Joystick Control Teleoperation
- [ ] Realtime Arm Servoing
- [ ] Benchmarking
- [ ] Integration/Unit Tests
- [ ] Debugging Tests

I also found the following changes that need to be performed to the [moveit_tutorials](https://github.com/ros-planning/moveit_tutorials):

- [ ]  [setup_assistant_tutorial.html](https://ros-planning.github.io/moveit_tutorials/doc/setup_assistant/setup_assistant_tutorial.html):
    -  [ ] Should have updated pictures (i.e., menu order was changed).
    -  [ ] Panda description file was changed to be `/franka_description/robots/panda.urdf.xacro` the hand is now enabled by using the `hand:=true` xacro parameter.
    -  [ ] The virtual joint should not be added to the `panda_arm` group as it causes problems with the Gazebo simulation (see https://github.com/rhaschke/panda_moveit_config/pull/8#discussion_r733093022).
- [ ] Maybe the `panda_control_moveit_rviz.launch` file should be documented in the documentation.
- [ ] The `demo_chomp.launch` file was added to the [panda_moveit_config](https://github.com/ros-planning/panda_moveit_config/) repository, but this file is not used in the moveit_tutorials. 
- [ ] The lerp planner documentation talks about a `lerp_example.launch` file but this file is not included I think this should be changed to refer the `demo_lerp.launch` file.
- [ ] The [gazebo simulation intergration tutorial](https://ros-planning.github.io/moveit_tutorials/doc/gazebo_simulation/gazebo_simulation.html) can be removed.

rhaschke, 2021-11-10:
I suggest opening a new issue for the moveit_tutorial tests which we can then reference to from here.

rickstaa, 2021-11-11:
The planners have been added to MSA in https://github.com/ros-planning/moveit/pull/2955.

rickstaa, 2021-11-17:
@rhaschke Can you update:

> Rework hand collision frankaemika/franka_ros#189
defining detailed (panda_hand) and coarse (panda_hand_coarse) collision models for environmental and self collisions resp.

To reflect the new pull request (i.e. https://github.com/frankaemika/franka_ros/pull/199).

rickstaa, 2022-01-29:
@rhaschke, I think some of the todos in this issue can be checked as done. For testing the tutorials, I'm waiting for @frankaemika to decide on https://github.com/frankaemika/franka_ros/pull/188 since if I do it now, things change I might have to retest.

gollth, 2022-02-02:
Hey there,

I'm testing out the new `noetic-devel` branch of this repo together with the new MoveIT and ran into some problem:

Suppose I want to start the move group in a launch file like so:

```xml
<?xml version="1.0" ?>
<launch>
  <arg name="robot_ip" />

  <arg name="transmission" value="position" />

  <include file="$(find franka_control)/launch/franka_control.launch" pass_all_args="true" />

  <include file="$(find panda_moveit_config)/launch/move_group.launch">

    <arg name="load_gripper" value="true" />
  </include>

</launch>
```

I get the error message to please set the `transmission` argument:

```
included file [/home/franka/ros/src/panda_moveit_config/launch/simple_moveit_controller_manager.launch.xml] requires the 'transmission' arg to be set
```

However the `move_group.launch` does not accept such an argument:

```
RLException: unused args [transmission] for include of [/home/franka/ros/src/panda_moveit_config/launch/move_group.launch]
```

AFAIR in Melodic you didn't need to (and couldn't) provide an extra "transmission" argument to the `move_group.launch`.

How is this indented to work now?

rhaschke, 2022-02-02:
In `demo.launch` it seems to work because we include `move_group.launch` with `pass_all_args="true"`:
https://github.com/ros-planning/panda_moveit_config/blob/c38a59af4ca94c008b3677e1ee1787a64bc6fea3/launch/demo.launch#L55

There is a working example file very similar to your example: [franka_control.launch](https://github.com/ros-planning/panda_moveit_config/blob/c38a59af4ca94c008b3677e1ee1787a64bc6fea3/launch/franka_control.launch).

gollth, 2022-02-03:
Thanks for the answer, I thought so as well that this is the background. But this would mean that you cannot even use this launch file like this anymore:

```xml
<launch>
  <include file="$(find panda_moveit_config)/launch/move_group.launch" />
</launch>
```
I think many users of this repo will have something similar to this in their user code. Is this breaking change really something you intent? If yes is it already documented somewhere already?

rhaschke, 2022-02-03:
I will try to improve the launch files over the weekend.

gollth, 2022-03-09:
Hey @rhaschke any updates on this? Currently `panda_moveit_config` is in a not very usable state

rickstaa, 2022-08-09:
I had some questions I my email about how to use the new noetic branch so I will put this here as a reference:

1. Install ROS noetic (see http://wiki.ros.org/noetic/Installation/Ubuntu).
2. Build libfranka from source (see https://frankaemika.github.io/docs/installation_linux.html).
3. Build franka_ros from source (see https://frankaemika.github.io/docs/installation_linux.html).
4. Clone the noetic branch of [panda_moveit_config](https://github.com/ros-planning/panda_moveit_config/tree/noetic-devel).
5. Install the ros dependencies using:
```rosdep install --from-paths src --ignore-src --rosdistro kinetic -y --skip-keys libfranka````
6. Build the catkin workspace.
```catkin build -DCMAKE_BUILD_TYPE=Release -DFranka_DIR:PATH=<PATH_TO_LIBFRANKA>/libfranka/build```
7. Run the [ros-planning/panda_moveit_config/blob/noetic-devel/launch/demo_gazebo.launch](https://github.com/ros-planning/panda_moveit_config/blob/noetic-devel/launch/demo_gazebo.launch) launch file.
8. Move the robot using the RVIZ Moveit Planning GUI (see video).

https://user-images.githubusercontent.com/17570430/183608713-85fc98a9-5426-4467-8f47-0a629629bab9.mp4

rickstaa, 2022-08-09:
@rhaschke @gollth, I don't think anything is blocking us from releasing the noetic package. To my knowledge, the last time I checked, there were only the instabilities in the gripper action (gripper action time out, gripper action not working when the gripper is vertical - see https://github.com/frankaemika/franka_ros/issues/172).

rhaschke, 2022-08-09:
I agree. However, before releasing `panda_moveit_config`, `franka_ros` should be released including https://github.com/frankaemika/franka_ros/issues/199, which was merged in April. Also, we will diverge between Melodic and Noetic as MoveIt Melodic doesn't support the new collision matrix specification, i.e. coarse vs. detailed collision models required by `franka_description`. @gollth, are you fine with this?

gollth, 2022-08-29:
Just as a heads up, @marcbone is currently working on a URDF changing feature in `franka_ros`. The outer interface (i.e. XACRO args and filenames ...) will stay the same, but the internal structure will be refactored a bit. We don't think this will be a breaking change for `panda_moveit_config` but with URDFs you can never know how breaking it really is.

rhaschke, 2022-08-29:
> @marcbone is currently working on a URDF changing feature in `franka_ros`

Please inform me if this gets merged and I will have a look at `panda_moveit_config`.

gollth, 2022-08-30:
`franka_ros` 0.9.1 is now released on [melodic](https://github.com/ros/rosdistro/pull/34241) and [noetic](https://github.com/ros/rosdistro/pull/34242)

rhaschke, 2022-09-01:
I triggered a release into Noetic: https://github.com/ros/rosdistro/pull/34290

closed 2022-09-01 by rhaschke

marcbone, 2022-09-05:
> > @marcbone is currently working on a URDF changing feature in `franka_ros`
> 
> Please inform me if this gets merged and I will have a look at `panda_moveit_config`.

I guess there was a bit of miscommunication. The refactoring of the URDF is now done and on the develop branch, but it was not part of the`0.9.1` release. We will make a release over rosdistro for version `0.10.0` tomorrow.