moveit/panda_moveit_config #97: Update moveit_tutorials

rickstaa, 2021-11-10:
The MoveIt tutorials should be updated now that the new `noetic` branch has been released. This issue tracks this process.

## TODOS

@tylerjw already performed some tests for his solution with the [moveit_tutorials](https://github.com/ros-planning/panda_moveit_config/pull/74#issuecomment-703139739). We however need to re-perform these tests for the new Noetic release. Please first check https://github.com/ros-planning/panda_moveit_config/pull/74#issuecomment-703139739 before testing one of the following tutorials:

- [x] Getting Started
    - [ ] Works but two small pull requests are requried when the noetic-branch is ready: https://github.com/ros-planning/moveit_tutorials/pull/673 and https://github.com/ros-planning/moveit/pull/2924.
- [ ] MoveIt Quickstart in RViz
- [ ] Move Group C++ Interface
- [ ] Move Group Python Interface
- [ ] MoveIt Commander Scripting
- [ ] Robot Model and Robot State
- [ ] Planning Scene
- [ ] Planning Scene Monitor
- [ ] Planning Scene ROS API
- [ ] Motion Planning API
- [ ] Motion Planning Pipeline
- [ ] Creating MoveIt Plugins
- [ ] Visualizing Collisions
- [ ] Time Parameterization
- [ ] Planning with approximated Constraint Manifolds
- [ ] Pick and Place
- [ ] MoveIt Grasps
- [ ] MoveIt Task Constructor
- [ ] MoveIt Deep Grasps
- [ ] Subframes
- [ ] MoveItCpp Tutorial
- [ ] Using Bullet for Collision Checking
- [ ] Mesh Filter with UR5 and Kinect
- [ ] MoveIt Setup Assistant
    - [ ] The `panda_tool` will be renamed change group picture. 
    - [ ] The extended and ready pose in the pmc are in collision now. This should be fixed in the pmc and the pose picture should be updated.
- [ ] URDF and SRDF
- [ ] Low Level Controllers
- [ ] Preception Pipeline Tutorial
- [ ] IKFast Kinematics Solver
- [ ] TRACK-IK Kinematics Solver
- [ ] OPW Kinematics Solver for Industrial Manipulators
- [ ] Kinematics Configuration
- [ ] Custom Constraint Samplers
- [ ] Representation and Evaluation of Constraints
- [ ] OMPL Planner
- [ ] CHOMP Planner
- [ ] STOMP Planner
- [ ] TrajOpt Planner
- [ ] Pilz Industrial Motion Planner
- [ ] Sequence of multiple segments
- [ ] Planning Adapter Tutorials
- [ ] Warehouse - Persistent Scenes and States
- [ ] Joystick Control Teleoperation
- [ ] Realtime Arm Servoing
- [ ] Benchmarking
- [ ] Integration/Unit Tests
- [ ] Debugging Tests

I also found the following changes that need to be performed to the [moveit_tutorials](https://github.com/ros-planning/moveit_tutorials):

- [ ]  [setup_assistant_tutorial.html](https://ros-planning.github.io/moveit_tutorials/doc/setup_assistant/setup_assistant_tutorial.html):
    -  [ ] Should have updated pictures (i.e., menu order was changed).
    -  [ ] Panda description file was changed to be `/franka_description/robots/panda.urdf.xacro` the hand is now enabled by using the `hand:=true` xacro parameter.
    -  [ ] The virtual joint should not be added to the `panda_arm` group as it causes problems with the Gazebo simulation (see https://github.com/rhaschke/panda_moveit_config/pull/8#discussion_r733093022).
- [ ] Maybe the `panda_control_moveit_rviz.launch` file should be documented in the documentation.
- [ ] The `demo_chomp.launch` file was added to the [panda_moveit_config](https://github.com/ros-planning/panda_moveit_config/) repository, but this file is not used in the moveit_tutorials. 
- [ ] The lerp planner documentation talks about a `lerp_example.launch` file but this file is not included I think this should be changed to refer the `demo_lerp.launch` file.
- [ ] The [gazebo simulation intergration tutorial](https://ros-planning.github.io/moveit_tutorials/doc/gazebo_simulation/gazebo_simulation.html) can be removed.

rickstaa, 2021-11-10:
@rhaschke I started updating the [moveit setup assistant](https://github.com/ros-planning/moveit_tutorials) tutorial (see https://github.com/ros-planning/moveit_tutorials/pull/678). I, however, noticed that the poses used in the `panda_moveit_config` are now in a collision when I use the pull requests in https://github.com/ros-planning/panda_moveit_config/issues/96:

https://github.com/ros-planning/panda_moveit_config/blob/a831c092ea97789ece7d3c643c0ee367897e2e48/config/arm.xacro#L13-L30

I think this is because the MSA uses the coarse collisions (see https://github.com/ros-planning/moveit/blob/4db626d9c3b5d6296b012188a4a0bfe4bddf6bce/moveit_setup_assistant/src/widgets/robot_poses_widget.cpp#L785-L789). Should we choose new poses, or is this something we should fix in the MSA or franka_ros?

rickstaa, 2021-11-10:
Ah, I now see that https://github.com/rickstaa/msa_test mixed the results of the PMC with the newly created config. I will add a newly created robot configuration.

rhaschke, 2021-11-10:
The URDF description needs more rework: https://github.com/frankaemika/franka_ros/pull/199. But I don't have more time to invest in there...

rickstaa, 2021-11-11:
Thanks, I haven't seen that pull request yet. I will take a look.

arnauddomes, 2022-04-06:
Hello,
I try to run move_to_start.py example from franka_ros but on panda_moveit_config for noetic there is no file panda_moveit.launch.
Is there a way to make it fix that issue? Why that file has been removed?

rhaschke, 2022-04-06:
Please try `franka_control.launch` instead.

xav12358, 2022-04-06:
In fact I want to use an example with moveit. And in my move_to_start.launch there is already a call to franka_control.launch

```
<?xml version="1.0" ?>
<launch>
  <arg name="robot_ip" />

  <include file="$(find franka_control)/launch/franka_control.launch">
    <arg name="robot_ip" value="$(arg robot_ip)" />
    <arg name="load_gripper" value="false" />
  </include>
  <include file="$(find panda_moveit_config)/launch/panda_moveit.launch">
    <arg name="load_gripper" value="false" />
  </include>
  <node name="move_to_start" pkg="franka_example_controllers" type="move_to_start.py" output="screen" required="true" />
</launch>

```

 Is there an other example of Moveit ?

rhaschke, 2022-04-06:
Note that a `franka_control.launch` is provided by both, `franka_control` and `panda_moveit_config`. Just replace `panda_moveit.launch` with `franka_control.launch`.

xav12358, 2022-04-06:
After replacing panda_moveit.launch by franka_control.launch, I had to add robot_ip

```
<?xml version="1.0" ?>
<launch>
  <arg name="robot_ip" />

  <include file="$(find franka_control)/launch/franka_control.launch">
    <arg name="robot_ip" value="$(arg robot_ip)" />
    <arg name="load_gripper" value="false" />
  </include>
  <include file="$(find panda_moveit_config)/launch/franka_control.launch">
    <arg name="load_gripper" value="false" />
    <arg name="robot_ip" value="$(arg robot_ip)" />
  </include>
  <node name="move_to_start" pkg="franka_example_controllers" type="move_to_start.py" output="screen" required="true" />
</launch>

```

But I get that error:

arnauddomes, 2022-04-06:
After a try I get the same trouble as @xav12358 

I am on Ubuntu 20.04 and Noetic . And also Noetic on all my repos.

rhaschke, 2022-04-06:
Ok. Looking into `panda_moveit_config/franka_control.launch`, I see that this file already includes `franka_control/franka_control.launch`, which explains the error message.
I think, it should suffice to include move_group.launch:

```xml
<?xml version="1.0" ?>
<launch>
  <arg name="robot_ip" />

  <include file="$(find franka_control)/launch/franka_control.launch">
    <arg name="robot_ip" value="$(arg robot_ip)" />
    <arg name="load_gripper" value="false" />
  </include>
  <include file="$(find panda_moveit_config)/launch/move_group.launch">
    <arg name="load_gripper" value="false" />
  </include>
  <node name="move_to_start" pkg="franka_example_controllers" type="move_to_start.py" output="screen" required="true" />
</launch>
```

xav12358, 2022-04-06:
I copy paste your launch file and I get now that error:

```
[ERROR] [1649253426.041527992]: Exception while loading planner 'pilz_industrial_motion_planner::CommandPlanner': According to the loaded plugin descriptions the class pilz_industrial_motion_planner::CommandPlanner with base class type planning_interface::PlannerManager does not exist. Declared types are  chomp_interface/CHOMPPlanner ompl_interface/OMPLPlanner
Available plugins: chomp_interface/CHOMPPlanner, ompl_interface/OMPLPlanner
[ERROR] [1649253426.041559472]: Failed to initialize planning pipeline 'pilz_industrial_motion_planner'.
```

rhaschke, 2022-04-06:
@xav12358, looks like you are not running ROS Noetic, but ROS Melodic. In any case, you can ignore this error. It just states that the Pilz planner is not available (in Melodic).

xav12358, 2022-04-06:
I update my previous post. When I run the move_to_start.py the robot arm is not moving

xav12358, 2022-04-07:
Indeed I interrupted the program because no rviz was launch, just because it was not in the launch file. So I start rviz in an other terminal.

So now I get that full log:

```
 roslaunch franka_example_controllers move_to_start.launch robot_ip:=172.16.0.2
... logging to /home/dac/.ros/log/af27d34e-b644-11ec-9197-5f75f7ec2368/roslaunch-dac-6778.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dac:39555/

SUMMARY
========

PARAMETERS
 * /effort_joint_trajectory_controller/constraints/goal_time: 0.5
 * /effort_joint_trajectory_controller/constraints/panda_joint1/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint2/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint3/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint4/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint5/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint6/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint7/goal: 0.05
 * /effort_joint_trajectory_controller/gains/panda_joint1/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint1/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint1/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint2/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint2/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint2/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint3/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint3/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint3/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint4/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint4/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint4/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint5/d: 10
 * /effort_joint_trajectory_controller/gains/panda_joint5/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint5/p: 250
 * /effort_joint_trajectory_controller/gains/panda_joint6/d: 10
 * /effort_joint_trajectory_controller/gains/panda_joint6/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint6/p: 150
 * /effort_joint_trajectory_controller/gains/panda_joint7/d: 5
 * /effort_joint_trajectory_controller/gains/panda_joint7/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint7/p: 50
 * /effort_joint_trajectory_controller/joints: ['panda_joint1', ...
 * /effort_joint_trajectory_controller/type: effort_controller...
 * /franka_control/arm_id: panda
 * /franka_control/collision_config/lower_force_thresholds_acceleration: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/lower_force_thresholds_nominal: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/lower_torque_thresholds_acceleration: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/lower_torque_thresholds_nominal: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/upper_force_thresholds_acceleration: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/upper_force_thresholds_nominal: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/upper_torque_thresholds_acceleration: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/upper_torque_thresholds_nominal: [20.0, 20.0, 18.0...
 * /franka_control/cutoff_frequency: 100
 * /franka_control/internal_controller: joint_impedance
 * /franka_control/joint_limit_warning_threshold: 1
 * /franka_control/joint_names: ['panda_joint1', ...
 * /franka_control/rate_limiting: True
 * /franka_control/realtime_config: enforce
 * /franka_control/robot_ip: 172.16.0.2
 * /franka_state_controller/arm_id: panda
 * /franka_state_controller/joint_names: ['panda_joint1', ...
 * /franka_state_controller/publish_rate: 30
 * /franka_state_controller/type: franka_control/Fr...
 * /joint_state_publisher/rate: 30
 * /joint_state_publisher/source_list: ['franka_state_co...
[... 477 lines cut ...]
[ INFO] [1649316710.608907396]: ABORTED: Solution found but controller failed during execution

```

And the echo seems to work properly:
```

rostopic echo /move_group/status
header:
  seq: 58
  stamp:
    secs: 1649316965
    nsecs: 597271210
  frame_id: ''
status_list: []
---
header:
  seq: 59
  stamp:
    secs: 1649316965
    nsecs: 797292258
  frame_id: ''
status_list: []
---
```

xav12358, 2022-04-07:
I also try to launch the move_to_start.py script in an other terminal and I get that error:

rhaschke, 2022-04-07:
You are missing controllers:
```
[ERROR] [1649316710.575224828]: Unable to identify any set of controllers that can actuate the specified joints: [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 ]
[ERROR] [1649316710.575238790]: Known controllers and their joints:

```
because the corresponding controller_spawner fails:
```
[WARN] [1649316697.072075]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[panda/controller_spawner-6] process has finished cleanly
```
The reason for this is in `panda_moveit_config/ros_controllers.launch`, where the controller spawner is started within the namespace `ns="/$(arg arm_id)"`:
```xml
  <!-- Load and start the controllers -->
  <node ns="/$(arg arm_id)" name="controller_spawner" pkg="controller_manager" type="spawner" respawn="false" output="screen"
        args="$(arg transmission)_joint_trajectory_controller" />
```
The root cause of that problem is that @frankaemika uses a namespace (`arm_id = panda`) for Gazebo to run ROS control in, but doesn't use that namespace for the real robot. Please remove the namespace attribute.

xav12358, 2022-04-07:
Thank you. SO I remove in ros_controller.launch that line:```

  <node ns="/$(arg arm_id)" name="controller_spawner" pkg="controller_manager" type="spawner" respawn="false" output="screen"
        args="$(arg transmission)_joint_trajectory_controller" />
```

And I get that error:

```
[ERROR] [1649322376.865258245]: Unable to identify any set of controllers that can actuate the specified joints: [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 ]
[ERROR] [1649322376.865276988]: Known controllers and their joints:
controller 'franka_gripper' controls joints:
  panda_finger_joint1

[ERROR] [1649322376.865314632]: Apparently trajectory initialization failed
```

xav12358, 2022-04-07:
I also juste remove ns="/$(arg arm_id)" in ros_controller.launch

roslaunch franka_example_controllers move_to_start.launch robot_ip:=172.16.0.2
... logging to /home/dac/.ros/log/8e190288-b656-11ec-bcad-33cf265bcff1/roslaunch-dac-8793.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dac:36445/

SUMMARY
========

PARAMETERS
 * /effort_joint_trajectory_controller/constraints/goal_time: 0.5
 * /effort_joint_trajectory_controller/constraints/panda_joint1/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint2/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint3/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint4/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint5/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint6/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint7/goal: 0.05
 * /effort_joint_trajectory_controller/gains/panda_joint1/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint1/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint1/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint2/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint2/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint2/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint3/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint3/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint3/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint4/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint4/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint4/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint5/d: 10
 * /effort_joint_trajectory_controller/gains/panda_joint5/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint5/p: 250
 * /effort_joint_trajectory_controller/gains/panda_joint6/d: 10
 * /effort_joint_trajectory_controller/gains/panda_joint6/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint6/p: 150
 * /effort_joint_trajectory_controller/gains/panda_joint7/d: 5
 * /effort_joint_trajectory_controller/gains/panda_joint7/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint7/p: 50
 * /effort_joint_trajectory_controller/joints: ['panda_joint1', ...
 * /effort_joint_trajectory_controller/type: effort_controller...
 * /franka_control/arm_id: panda
 * /franka_control/collision_config/lower_force_thresholds_acceleration: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/lower_force_thresholds_nominal: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/lower_torque_thresholds_acceleration: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/lower_torque_thresholds_nominal: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/upper_force_thresholds_acceleration: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/upper_force_thresholds_nominal: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/upper_torque_thresholds_acceleration: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/upper_torque_thresholds_nominal: [20.0, 20.0, 18.0...
 * /franka_control/cutoff_frequency: 100
 * /franka_control/internal_controller: joint_impedance
 * /franka_control/joint_limit_warning_threshold: 1
 * /franka_control/joint_names: ['panda_joint1', ...
 * /franka_control/rate_limiting: True
 * /franka_control/realtime_config: enforce
 * /franka_control/robot_ip: 172.16.0.2
 * /franka_gripper/default_grasp_epsilon/inner: 0.005
 * /franka_gripper/default_grasp_epsilon/outer: 0.005
 * /franka_gripper/default_speed: 0.1
 * /franka_gripper/joint_names: ['panda_finger_jo...
 * /franka_gripper/publish_rate: 30
 * /franka_gripper/robot_ip: 172.16.0.2
 * /franka_gripper/stop_at_shutdown: False
 * /franka_state_controller/arm_id: panda
 * /franka_state_controller/joint_names: ['panda_joint1', ...
[... 379 lines cut ...]
You can start planning now!

[ WARN] [1649324362.372331076]: FrankaHW: 
	panda_joint2: 4.302033 degrees to joint limits (limits: [-1.762800, 1.762800] q: 1.687715) 
[ WARN] [1649324367.373192772]: FrankaHW: 
	panda_joint2: 4.302147 degrees to joint limits (limits: [-1.762800, 1.762800] q: 1.687713) 

[ WARN] [1649324372.374095409]: FrankaHW: 
	panda_joint2: 4.302033 degrees to joint limits (limits: [-1.762800, 1.762800] q: 1.687715) 
^C[move_group-8] killing on exit
[move_group/controller_spawner-7] killing on exit
[joint_state_publisher-6] killing on exit
[robot_state_publisher-5] killing on exit
[state_controller_spawner-4] killing on exit
[franka_control-3] killing on exit
[franka_gripper-2] killing on exit
[INFO] [1649324372.632870]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1649324372.634642]: Stopping all controllers...
^C[WARN] [1649324372.827723]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 104] Connection reset by peer
^C^C^C[WARN] [1649324373.237522]: Controller Spawner couldn't find the expected controller_manager ROS interface.
^C^C[rosout-1] killing on exit
^C^C[master] killing on exit
^Cshutting down processing monitor...
... shutting down processing monitor complete
done

rhaschke, 2022-04-07:
Controllers still seem not active yet. Please check controller status via http://wiki.ros.org/rqt_controller_manager and http://wiki.ros.org/rqt_joint_trajectory_controller.

xav12358, 2022-04-07:
Here is the result:

[image]

rhaschke, 2022-04-07:
Please start the desired controller manually. I don‘t understand, why the controller isn‘t started by the controller spawner in ros_controllers.launch. You should investigate this command.

arnauddomes, 2022-04-07:
What could I find the list of controller? and how could I launch it?

xav12358, 2022-04-07:
Is it roslaunch that example?
`
roslaunch franka_example_controllers cartesian_pose_example_controller.launch robot_ip:=172.16.0.2 debug:=true`

rhaschke, 2022-04-07:
@arnauddomes and @xav12358, please open a _new_ issue to continue the conversation. You kind of hijacked this thread...

xav12358, 2022-04-07:
Ok I will create a new issue

rickstaa, 2023-11-21:
@rhaschke This issue is also being tracked at https://github.com/ros-planning/moveit_tutorials/issues/525. Let's close it here?

rhaschke, 2023-11-21:
Sure.

closed 2023-11-21 by rhaschke