moveit/panda_moveit_config #100: roslaunch demo.launch fails

orrblue, 2022-02-07:
I'm on Ubuntu 20.04 + ROS Noetic and I downloaded the noetic-devel branch of this repo, but when I try `roslaunch panda_moveit_config demo.launch` there is an error caused by the file `panda_moveit_config/config/panda_arm.srdf.xacro` (i think). However when I clone the melodic-devel branch, the error is gone. I understand the noetic-devel branch is still in development, but I wanted to highlight this issue. The error I get is pasted below:

`setting /run_id to 1a3f64ec-884d-11ec-9672-29c550d0c8eb
process[rosout-1]: started with pid [243997]
started core service [/rosout]
process[virtual_joint_broadcaster_0-2]: started with pid [244004]
process[joint_state_publisher-3]: started with pid [244005]
process[robot_state_publisher-4]: started with pid [244006]
process[move_group-5]: started with pid [244012]
process[rviz_nitz_243975_6914526873867292394-6]: started with pid [244013]
[ERROR] [1644262529.644142658]: Link 'panda_hand_tcp' declared as part of a chain in group 'manipulator' is not known to the URDF
[ WARN] [1644262529.645752808]: Group 'manipulator' is empty.
[ERROR] [1644262529.646011577]: Link 'panda_hand_tcp' specified as parent for end effector 'hand_tcp' is not known to the URDF
[ WARN] [1644262529.646079339]: Link 'panda_link0_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.646100600]: Link 'panda_link0_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.646114909]: Link 'panda_link0_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.646129192]: Link 'panda_link0_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.646141848]: Link 'panda_link0_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.646155262]: Link 'panda_link0_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.646168350]: Link 'panda_link0_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.646181205]: Link 'panda_link0_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.646194263]: Link 'panda_link0_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.646224033]: Link 'panda_link1_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.646257751]: Link 'panda_link1_sc' is not known to URDF. Cannot disable collisons.
.
.
.
[ WARN] [1644262529.647164256]: Link 'panda_link7_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.647186457]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.647200254]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.647213326]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.647226745]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.647240364]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.647256477]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.647270295]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.647284219]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.647296753]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.647333173]: Link 'panda_hand_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.647361510]: Link 'panda_hand_sc' is not known to URDF. Cannot disable collisons.
[ INFO] [1644262529.647537810]: Loading robot model 'panda'...
[ WARN] [1644262529.647744382]: Link panda_leftfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1644262529.647787841]: Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1644262529.648011708]: Group 'manipulator' must have at least one valid joint
[ WARN] [1644262529.648044860]: Failed to add group 'manipulator'
[ERROR] [1644262529.648169000]: Group state 'ready' specified for group 'manipulator', but that group does not exist
[ERROR] [1644262529.648193135]: Group state 'extended' specified for group 'manipulator', but that group does not exist
[ INFO] [1644262529.738328044]: Publishing maintained planning scene on 'monitored_planning_scene'
[ INFO] [1644262529.741363760]: Listening to 'joint_states' for joint states
[ INFO] [1644262529.746310131]: Listening to '/attached_collision_object' for attached collision objects
[ INFO] [1644262529.746410625]: Starting planning scene monitor
[ INFO] [1644262529.748972487]: Listening to '/planning_scene'
[ INFO] [1644262529.749039179]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[ INFO] [1644262529.754676352]: Listening to '/collision_object'
[ INFO] [1644262529.756960791]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1644262529.757859545]: No 3D sensor plugin(s) defined for octomap updates
[ INFO] [1644262529.759127868]: Loading planning pipeline 'chomp'
[ERROR] [1644262529.761756215]: Exception while loading planner 'chomp_interface/CHOMPPlanner': According to the loaded plugin descriptions the class chomp_interface/CHOMPPlanner with base class type planning_interface::PlannerManager does not exist. Declared types are  lerp_interface/LERPPlanner ompl_interface/OMPLPlanner
Available plugins: lerp_interface/LERPPlanner, ompl_interface/OMPLPlanner
[ INFO] [1644262529.766224328]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1644262529.766836441]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1644262529.767328463]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1644262529.767855585]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1644262529.768421041]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1644262529.769237613]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1644262529.769351337]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1644262529.769398999]: Using planning request adapter 'Resolve constraint frames to robot links'
[ INFO] [1644262529.769431178]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1644262529.769461905]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1644262529.769496618]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1644262529.769527450]: Using planning request adapter 'Fix Start State Path Constraints'
[... 39 lines cut ...]
.
[ WARN] [1644262529.880348976]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.880365284]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.880381794]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.880398601]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.880416034]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.880432575]: Link 'panda_link8_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.880474292]: Link 'panda_hand_sc' is not known to URDF. Cannot disable collisons.
[ WARN] [1644262529.880502356]: Link 'panda_hand_sc' is not known to URDF. Cannot disable collisons.
[ INFO] [1644262529.880574724]: Loading robot model 'panda'...
[ WARN] [1644262529.880694955]: Link panda_leftfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1644262529.880723390]: Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1644262529.880812374]: Group 'manipulator' must have at least one valid joint
[ WARN] [1644262529.880833247]: Failed to add group 'manipulator'
[ERROR] [1644262529.880913331]: Group state 'ready' specified for group 'manipulator', but that group does not exist
[ERROR] [1644262529.880934594]: Group state 'extended' specified for group 'manipulator', but that group does not exist
[... cut ...]
[ INFO] [1644262529.921683644]: Fake controller 'fake_hand_controller' with joints [ panda_finger_joint1 ]
[ INFO] [1644262529.921961756]: Fake controller 'fake_panda_arm_hand_controller' with joints [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 panda_finger_joint1 ]
[ INFO] [1644262529.922285877]: Returned 3 controllers in list
[ INFO] [1644262529.934815681]: Trajectory execution is managing controllers
[ INFO] [1644262529.934855674]: MoveGroup debug mode is ON
`

simonschmeisser, 2022-02-07:
Please use tripple backticks for multiline logs/code

rhaschke, 2022-02-07:
As you mentioned, the noetic-devel branch is still work-in-progress. Particularly, you need to use a modified version of `franka_description`. See https://github.com/ros-planning/panda_moveit_config/issues/96 for details.

closed 2022-02-07 by rhaschke