moveit/panda_moveit_config #107: Error when I use move_to_start

xav12358, 2022-04-07:
hello

Following the previous discussion on that issue:
https://github.com/ros-planning/panda_moveit_config/issues/97

I work on Noetic on Ubuntu 20.04. I have trouble to use move_to_start.launch from franka_ros. I modified the move_to_start.launch file according to @rhaschke advice.

```
<?xml version="1.0" ?>
<launch>
  <arg name="robot_ip" />

  <include file="$(find franka_control)/launch/franka_control.launch">
    <arg name="robot_ip" value="$(arg robot_ip)" />
    <arg name="load_gripper" value="false" />
  </include>
  <include file="$(find panda_moveit_config)/launch/move_group.launch">
    <arg name="load_gripper" value="false" />
  </include>
  <node name="move_to_start" pkg="franka_example_controllers" type="move_to_start.py" output="screen" required="true" />
</launch>

```

I get that log:

```
roslaunch franka_example_controllers move_to_start.launch robot_ip:=172.16.0.2 debug:=true
... logging to /home/dac/.ros/log/601ef3f6-b66f-11ec-bcad-33cf265bcff1/roslaunch-dac-19458.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dac:42491/

SUMMARY
========

PARAMETERS
 * /effort_joint_trajectory_controller/constraints/goal_time: 0.5
 * /effort_joint_trajectory_controller/constraints/panda_joint1/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint2/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint3/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint4/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint5/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint6/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint7/goal: 0.05
 * /effort_joint_trajectory_controller/gains/panda_joint1/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint1/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint1/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint2/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint2/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint2/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint3/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint3/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint3/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint4/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint4/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint4/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint5/d: 10
 * /effort_joint_trajectory_controller/gains/panda_joint5/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint5/p: 250
 * /effort_joint_trajectory_controller/gains/panda_joint6/d: 10
 * /effort_joint_trajectory_controller/gains/panda_joint6/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint6/p: 150
 * /effort_joint_trajectory_controller/gains/panda_joint7/d: 5
 * /effort_joint_trajectory_controller/gains/panda_joint7/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint7/p: 50
 * /effort_joint_trajectory_controller/joints: ['panda_joint1', ...
 * /effort_joint_trajectory_controller/type: effort_controller...
[... 444 lines cut ...]
[ERROR] [1649335038.072478020]: Known controllers and their joints:

[ERROR] [1649335038.072503930]: Apparently trajectory initialization failed
[ INFO] [1649335038.072742030]: ABORTED: Solution found but controller failed during execution
================================================================================REQUIRED process [move_to_start-8] has died!
process has finished cleanly
log file: /home/dac/.ros/log/601ef3f6-b66f-11ec-bcad-33cf265bcff1/move_to_start-8*.log
Initiating shutdown!
================================================================================
[move_to_start-8] killing on exit
[move_group-7] killing on exit
[robot_state_publisher-4] killing on exit
[state_controller_spawner-3] killing on exit
[franka_control-2] killing on exit
[joint_state_publisher-5] killing on exit
[INFO] [1649335039.122425]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1649335039.123570]: Stopping all controllers...
[WARN] [1649335039.240186]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 104] Connection reset by peer
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

arnauddomes, 2022-04-07:
I have modify the move_to_start_launch file like this:

```
<?xml version="1.0" ?>
<launch>
  <arg name="robot_ip" />

  <include file="$(find franka_control)/launch/franka_control.launch">
    <arg name="robot_ip" value="$(arg robot_ip)" />
    <arg name="load_gripper" value="false" />
  </include>
  <include file="$(find panda_moveit_config)/launch/move_group.launch">
    <arg name="load_gripper" value="true" />
  </include>
 
</launch>
```

And then run the move_to_start.py in an other terminal.
In the first terminal I get that error:

```
roslaunch franka_example_controllers move_to_start.launch robot_ip:=172.16.0.2 debug:=true
... logging to /home/arnaud/.ros/log/cd8188ba-b678-11ec-bcad-33cf265bcff1/roslaunch-arnaud-25310.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://arnaud:39745/

SUMMARY
========

PARAMETERS
 * /effort_joint_trajectory_controller/constraints/goal_time: 0.5
 * /effort_joint_trajectory_controller/constraints/panda_joint1/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint2/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint3/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint4/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint5/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint6/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint7/goal: 0.05
 * /effort_joint_trajectory_controller/gains/panda_joint1/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint1/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint1/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint2/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint2/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint2/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint3/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint3/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint3/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint4/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint4/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint4/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint5/d: 10
 * /effort_joint_trajectory_controller/gains/panda_joint5/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint5/p: 250
 * /effort_joint_trajectory_controller/gains/panda_joint6/d: 10
 * /effort_joint_trajectory_controller/gains/panda_joint6/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint6/p: 150
 * /effort_joint_trajectory_controller/gains/panda_joint7/d: 5
 * /effort_joint_trajectory_controller/gains/panda_joint7/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint7/p: 50
 * /effort_joint_trajectory_controller/joints: ['panda_joint1', ...
 * /effort_joint_trajectory_controller/type: effort_controller...
 * /franka_control/arm_id: panda
 * /franka_control/collision_config/lower_force_thresholds_acceleration: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/lower_force_thresholds_nominal: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/lower_torque_thresholds_acceleration: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/lower_torque_thresholds_nominal: [20.0, 20.0, 18.0...
[... 432 lines cut ...]
[ WARN] [1649339251.739121864]: Link 'panda_link0_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739128870]: Link 'panda_link0_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739135359]: Link 'panda_link0_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739141608]: Link 'panda_link0_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739148231]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739154697]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739161370]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739167612]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739174273]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739180786]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739187176]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739193205]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739199643]: Link 'panda_link3_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739206205]: Link 'panda_link3_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739213296]: Link 'panda_hand_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739219697]: Link 'panda_hand_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739225923]: Link 'panda_hand_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1649339251.739232452]: Link 'panda_hand_sc' is not known to URDF. Cannot disable/enable collisons.
[ INFO] [1649339251.739336531]: Loading robot model 'panda'...
[... cut ...] ometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1649339251.739411965]: Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ INFO] [1649339252.782120718]: Ready to take commands for planning group panda_arm.
[ INFO] [1649339252.825843862]: ABORTED: Solution found but controller failed during execution

```

arnauddomes, 2022-04-07:
In fact the effort_joint_trajectory_controller and  position_joint_trajectory_controller are only generate with franka_ros default_controllers.yaml file but this file is never used in move_group.launch

arnauddomes, 2022-04-08:
I try that fix:
https://github.com/ros-planning/panda_moveit_config/pull/108/files

But the panda_moveit_config/config/ros_controller.yaml is empty.

arnauddomes, 2022-04-08:
Here is the diff I made to make it work

xav12358, 2022-04-08:
It seems it worls for me also. Thank you @arnauddomes

closed 2022-04-10 by rhaschke