moveit/panda_moveit_config #110: What version do we get when we do "sudo apt-get install ros-noetic-panda-moveit-config"?

Olimoyo, 2022-05-20:
I struggled quite a lot to get this repo to work with the `move_to_start.launch` example of the noetic-devel branch of franka_ros: https://github.com/frankaemika/franka_ros/blob/noetic-devel/franka_example_controllers/launch/move_to_start.launch.

The one set up that worked for me was to get the debian version of `ros-noetic-panda-moveit-config` and to remove the remapping in `move_group.launch`:

```
<remap from="/joint_states" to="/joint_states_desired" />
```

Is it possible to reproduce this setup with source files ?

BarisYazici, 2022-06-21:
Answering your question in the title:
1) You can see that when you are installing the package
`Get:1 http://packages.ros.org/ros/ubuntu focal/main amd64 ros-noetic-panda-moveit-config amd64 0.7.5-1focal.20220515.082357 [20,4 kB] `
2) You can also check the package.xml in the package directory
`roscd panda_moveit_config`
`cat package.xml`

rhaschke, 2022-08-23:
The debian package `ros-noetic-panda-moveit-config` is tagged with version 0.7.5
(check e.g. with `dpkg -l ros-noetic-panda-moveit-config`).
Unfortunately, the noetic branches of franka_ros and panda_moveit_config are currently out of sync. We are working on it...

closed 2022-08-23 by Olimoyo