moveit/panda_moveit_config #112: TIMELY: Cannot control gripper in Noetic version

harryzhangOG, 2022-05-30:
Hi I just installed the noetic version on Ubuntu 20 and was able to control a real robot arm using moveit and python binding. However, I could not close the gripper... Here is my step-to-step illustration of the issue:

First, here is the python code that runs well on the melodic version:
```Python
import rospy
import actionlib
from control_msgs.msg import GripperCommandAction, GripperCommandGoal

"""
This file tests the gripper (open <--> close)
"""

def main():
    rospy.init_node("grippter_test")
    # breakpoint()
    gripper_client = actionlib.SimpleActionClient('/franka_gripper/gripper_action', GripperCommandAction)
    gripper_client.wait_for_server()
    rospy.loginfo('gripper action server started.')
    goal = GripperCommandGoal()
    goal.command.position = 0
    # goal.width = 0.05
    gripper_client.send_goal(goal)
    rospy.loginfo('goal sent.')
    gripper_client.wait_for_result()

if __name__=="__main__":
    main()
```
Running the script would be stuck at `gripper_client.wait_for_result()`.

Then, in rviz, if i choose the planning group as `panda_hand`, I can see the gripper visualization getting closed:
[image]
However, the real robot does not move and I get the following error in the console:
```shell
[ INFO] [1653884561.353749456]: SimpleSetup: Path simplification took 0.047645 seconds and changed from 7 to 2 states
[ INFO] [1653884561.354453863]: Planning adapters have added states at index positions: [ 0 ]
[ERROR] [1653884566.691940361]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 5.314105 seconds). Stopping trajectory.
[ INFO] [1653884566.691990321]: Cancelling execution for franka_gripper
[ INFO] [1653884566.692089448]: Completed trajectory execution with status TIMED_OUT ...
[ INFO] [1653884566.728909471]: ABORTED: Timeout reached
```
Then I checked which controllers are running via: `rosservice call /controller_manager/list_controllers` and here is the output, no gripper controller is getting run...
```
controller: 
  - 
    name: "franka_state_controller"
    state: "running"
    type: "franka_control/FrankaStateController"
    claimed_resources: 
      - 
        hardware_interface: "franka_hw::FrankaStateInterface"
        resources: []
  - 
    name: "position_joint_trajectory_controller"
    state: "running"
    type: "position_controllers/JointTrajectoryController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::PositionJointInterface"
        resources: 
          - panda_joint1
          - panda_joint2
          - panda_joint3
          - panda_joint4
          - panda_joint5
          - panda_joint6
          - panda_joint7
```
Finally, I wrote this controller yaml file for the default controller:
```yaml
 controller_list:
  - name: position_joint_trajectory_controller
    action_ns: follow_joint_trajectory
    type: FollowJointTrajectory
    default: true
    joints:
      - panda_joint1
      - panda_joint2
      - panda_joint3
      - panda_joint4
      - panda_joint5
      - panda_joint6
      - panda_joint7
  - name: franka_gripper
    action_ns: gripper_action
    type: GripperCommand
    default: true
    joints:
      - panda_finger_joint1
      - panda_finger_joint2
```
`
Could anyone provide some pointers on how to fix this? This is really urgent so any feedback would be greatly appreciated. Thanks.

AndyZe, 2022-05-30:
There are a few parts to this:

- Add the gripper controller to the `moveit controllers yaml` -- looks like you've already done this.
- Add the gripper controller to the `ros_control yaml` -- maybe that has not been done.
- Add the gripper controller to the launch file.

Here are some examples:
- `ros_control yaml`:   https://github.com/ros-planning/moveit_resources/blob/master/panda_moveit_config/config/ros_controllers.yaml
- `ros_control launching`: https://github.com/ros-planning/moveit_resources/blob/master/panda_moveit_config/launch/simple_moveit_controller_manager.launch.xml

rhaschke, 2022-06-03:
The listed python code works well for me on Noetic as well. By the way, this code is not MoveIt related at all. It just requires franka_control to be running: `roslaunch franka_control franka_control.launch`.

rhaschke, 2022-09-01:
As the issue is not reproducible (anymore?), I'm closing the issue due to inactivity.

closed 2022-09-01 by rhaschke