moveit/panda_moveit_config #115: How to control a real franka robot arm?

jnumingdeli, 2022-07-17:
I want to use moveit to control a real franka robot arm. I try changing the value of fake_execution to false in demo.launch, but it doesn't work. and terminal shows "Action client not connected: position_joint_trajectory_controller/follow_joint_trajectory". 

What should I do?  Thank you very much.

rhaschke, 2022-07-17:
Obviously, you need to start your robot hardware controllers. In branch noetic-devel an example launch file exists to do so: [`franka_control.launch`](https://github.com/ros-planning/panda_moveit_config/blob/noetic-devel/launch/franka_control.launch)

jnumingdeli, 2022-07-18:
> Obviously, you need to start your robot hardware controllers. In branch noetic-devel an example launch file exists to do so: [`franka_control.launch`](https://github.com/ros-planning/panda_moveit_config/blob/noetic-devel/launch/franka_control.launch)

Thank you very much!!! It works. Now moveit can successfully drive the real robot arm.

closed 2022-07-18 by rhaschke

cnaes93, 2022-09-07:
Hi, I want to do the same thing, control the real robot with moveit. I changed the moveit_controller_manger to ros_control and also fake_execution to false but the real robot still does not follow the simulation path. which oder things should I change to make the real robot move?
thanks

rhaschke, 2022-09-07:
@cnaes93, please provide detailed info how you launch the robot and MoveIt. As said above, it should suffice to use `roslaunch panda_moveit_config franka_control.launch` using the latest releases of panda_moveit_config and franka_ros in ROS Noetic.
If that doesn't work, please check the error messages on the console. They provide useful hints.

cnaes93, 2022-09-13:
@rhaschke , firstof all I copied the panda_control_moveit_rviz.launch from github in the folder panda_moveit_config, because there is no such files in [ noetic version of moveit.](https://github.com/ros-planning/panda_moveit_config/tree/noetic-devel/launch) then I changed the fake controller in roscontrol as mentioned above and launch it. I also tried to launched the franka_control as you said, but it can not find the source of franka_control. ( I also sourced it like: `source ~/ws_moveit/devel/setup.bash` ) (I added in the~/.bashrc the following sources: `source /opt/ros/noetic/setup.bash
source ~/catkin_ws/devel/setup.bash        source ~/ws_moveit/devel/setup.bash`
[image]
[image]
How can I define the source-files to it and solve the problem?

rhaschke, 2022-09-13:
Did you install the debian package `ros-noetic-franka-control`? That's needed for your real robot!

cnaes93, 2022-09-14:
I installed the debian Package as well (I followed all the installations steps as mentioned in moveit noetic).
One time i did `catkin clean --deinit` my home directory since then it could not find any Libraries or packages. I reinstalled after that all ROS, libfranka, franka_ros and moveit but it did not solve the problem! Do you think its because of this or something else?

rhaschke, 2022-09-14:
Obviously, the package `franka_control` is not found. Hence, I can only conclude that you didn't install it.
Search for it in /opt/ros/noetic to confirm: `find /opt/ros/noetic -iname franka_control`.

cnaes93, 2022-09-14:
Thanks. The franka-control was not installed I don't know how but I installed it with: 
`sudo apt install ros-noetic-franka-control` and it works now.
Thanks.