moveit/panda_moveit_config #118: collision problem from start the panda moveit demo.launch in noetic version

cnaes93, 2022-08-19:
Hi
I installed recently the moveit for noetic from [link](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html) and clone the panda-moveit.config the melodic version (-b melodic-devel) like in the Installation was. But when I launch it 
`roslaunch panda_moveit_config demo.launch rviz_tutorial:=true`
 from the start the panda robot is in collision state I think (the links are red) like in the picture.
[image]

Then I reinstalled it this time with (-b noetic-devel). It didn't solved the Problem. 
 After that i tried to do a motion planning but I got the error:
**Panda-arm:Motion planning start tree could not be initialized**

How can I solve this issue?

rhaschke, 2022-08-19:
The melodic-devel branch of panda_moveit_config is known to have collision issues. However, this was solved (but not yet released) in Noetic. Hence, running the most recent code (of MoveIt@master and panda_moveit_config@noetic-devel) this works for me:
[image]

Could you please check/specify the exact versions of MoveIt and panda_moveit_config you use?
Given that your group is called hand (instead of panda_hand), I suspect that you are using old versions.

cnaes93, 2022-08-19:
How can I find the version of moveit and panda_moveit_config?

rhaschke, 2022-08-19:
> How can I find the version?

https://unix.stackexchange.com/questions/6284/how-do-i-check-package-version-using-apt-get-aptitude
https://stackoverflow.com/questions/949314/how-do-i-get-the-hash-for-the-current-commit-in-git

cnaes93, 2022-08-31:
I solved it. I started the installing  from [Getting Started ](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html) and there is no `sudo apt install ros-noetic-moveit`
so i actually didn't installed the moveit, but I'm wondering how could I opend the moveit without installing it. How ever I find my problem after I compared the installings code between Melodic and Notic version. There is no sudo apt install ros-notic-moveit in noetic version (it's in first page of moveit, not in getting started).

closed 2022-08-31 by cnaes93

cnaes93, 2022-08-31:
I did the installation via rosdep as explained there on 2 devices but there was no Panda_hand in MotionPlanning (just hand) and collision problem for start planning the robot (Panda-arm:Motion planning start tree could not be initialized). So reinstalled moveit again this time with  `sudo apt install ros-noetic-moveit` and then installing the packages and finally evrythings work well.