moveit/panda_moveit_config #120: No such file or directory:

YXGuan, 2022-09-11:
Hi there, 

I update my local [panda_moveit_config](https://github.com/ros-planning/panda_moveit_config) repo today and received the following error message: 

[image]

rhaschke, 2022-09-11:
Please also update `franka_description`! It's released but not yet synced.

closed 2022-09-11 by rhaschke

YXGuan, 2022-09-12:
I see. Thank you sir

YXGuan, 2022-09-12:
@rhaschke  
Hi Dr. Haschke, 

Sorry I have a question unrelated to this issue:
When I was attempting to add a griper in Define Planning Groups in MoveIt Setup Assistant, I can't see any available links related to panda_hand, panda_leftfinger, and panda_rightfinger. Did I do something stupid or am I using the wrong URDF file?

Thanks, 

[image]

[image]

rhaschke, 2022-09-12:
In order to get a gripper, you need to pass xacro arguments: `hand:=true`

YXGuan, 2022-09-12:
Thank you very much