moveit/panda_moveit_config #123: Cannot find 'panda_control_moveit_rviz.launch'

fzy139, 2022-10-05:
I want to calibrate my camera using `easy_handeye`. In its example the robot is started by `panda_control_moveit_rviz.launch`. However, this file could not be found anymore. Maybe it was deleted in a commit, for I find this file in bing.
Is there any alternative to this file?

rhaschke, 2022-10-05:
This file was replaced by `franka_control.launch`. See https://github.com/IFL-CAMP/easy_handeye/pull/112 for a fix.

closed 2022-10-05 by rhaschke

fzy139, 2022-10-05:
I changed my `.launch` file as your commit to `easy_handeye`. But I met some errors:
```
[ERROR] [1664965972.593216330]: Group state 'ready' specified for group 'panda_manipulator', but that group does not exist
[ERROR] [1664965972.593241398]: Group state 'extended' specified for group 'panda_manipulator', but that group does not exist
[ INFO] [1664965972.646884641]: Starting planning scene monitor
[ INFO] [1664965972.648844193]: Listening to '/move_group/monitored_planning_scene'
[ERROR] [1664965972.655189737]: Link 'panda_hand_tcp' declared as part of a chain in group 'panda_manipulator' is not known to the URDF
[ WARN] [1664965972.655226545]: Group 'panda_manipulator' is empty.
[ERROR] [1664965972.655409215]: Link 'panda_hand_tcp' specified as parent for end effector 'panda_hand_tcp' is not known to the URDF
```
Is there some way to fix it? My ROS version is Noetic.

rhaschke, 2022-10-05:
Which version of panda_moveit_config and franka_description do you have installed? Latest are:
```
ros-noetic-panda-moveit-config                             0.8.1-1focal.20221003.150820
ros-noetic-franka-description                              0.10.1-1focal.20220926.212146
```

fzy139, 2022-10-06:
When I tried `roslaunch panda_moveit_config demo.launch`, I got the error 
```bash
$ roslaunch panda_moveit_config demo.launch 
... logging to /home/hyperplane/.ros/log/efe5f270-4556-11ed-965b-bf2b1c9e7cb8/roslaunch-hyperplane-98756.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt

No such file or directory: /home/hyperplane/ros-neotic-franka/catkin_ws/src/franka_ros/franka_description/robots/panda/panda.urdf.xacro [Errno 2] No such file or directory: '/home/hyperplane/ros-neotic-franka/catkin_ws/src/franka_ros/franka_description/robots/panda/panda.urdf.xacro'
RLException: while processing /opt/ros/noetic/share/panda_moveit_config/launch/move_group.launch:
while processing /opt/ros/noetic/share/panda_moveit_config/launch/planning_context.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [['xacro', '/home/hyperplane/ros-neotic-franka/catkin_ws/src/franka_ros/franka_description/robots/panda/panda.urdf.xacro', 'hand:=true', 'arm_id:=panda']] returned with code [2]. 

Param xml is <param name="$(arg robot_description)" command="xacro '$(find franka_description)/robots/panda/panda.urdf.xacro' hand:=$(arg load_gripper) arm_id:=$(arg arm_id)" if="$(arg load_robot_description)"/>
The traceback for the exception was written to the log file
```

fzy139, 2022-10-06:
I find no directory in `xxx/catkin_ws/src/franka_ros/franka_description/robots/`.But there are several `.xacro` files:
```
dual_panda_example.urdf.xacro  hand.xacro            panda_arm.xacro     utils.xacro
hand.urdf.xacro                panda_arm.urdf.xacro  panda_gazebo.xacro
```

rhaschke, 2022-10-06:
> `ros-noetic-desktop-full` _should_ contain the latest `ros-noetic-panda-moveit-config` and `ros-noetic-franka-description`.

Please explicitly check: `dpkg -l | grep <package name>`

> ```shell
> No such file or directory: /home/hyperplane/ros-neotic-franka/catkin_ws/src/franka_ros/franka_description/robots/panda/panda.urdf.xacro [Errno 2] No such file or directory: '/home/hyperplane/ros-neotic-franka/catkin_ws/src/franka_ros/franka_description/robots/panda/panda.urdf.xacro'
> ```

As you can see [here](https://github.com/frankaemika/franka_ros/tree/0.10.1/franka_description/robots/panda), `franka_description 0.10.1` comprises this file. Older versions don't! Please check your versions!

As you are building from source, you should `git remote update` and `git pull` the latest versions of `franka_ros` and `panda_moveit_config`!

fzy139, 2022-10-07:
Thanks for your answer. I updated my franka_ros and the error about `.xacro` files disappears. However, I got another errors when running `roslaunch panda_moveit_config demo.launch`:
```
[ INFO] [1665105587.197342563]: Loading planning pipeline 'chomp'
[ERROR] [1665105587.198660476]: Exception while loading planner 'chomp_interface/CHOMPPlanner': According to the loaded plugin descriptions the class chomp_interface/CHOMPPlanner with base class type planning_interface::PlannerManager does not exist. Declared types are  ompl_interface/OMPLPlanner
Available plugins: ompl_interface/OMPLPlanner
...
[ERROR] [1665105587.203065213]: Failed to initialize planning pipeline 'chomp'.
...
[ INFO] [1665105587.255082063]: Loading planning pipeline 'pilz_industrial_motion_planner'
[ERROR] [1665105587.256202775]: Exception while loading planner 'pilz_industrial_motion_planner::CommandPlanner': According to the loaded plugin descriptions the class pilz_industrial_motion_planner::CommandPlanner with base class type planning_interface::PlannerManager does not exist. Declared types are  ompl_interface/OMPLPlanner
Available plugins: ompl_interface/OMPLPlanner
[ERROR] [1665105587.256241003]: Failed to initialize planning pipeline 'pilz_industrial_motion_planner'.
```
I have checked that `ompl` is installed with ROS, but I cannot find `chomp`:
```bash
~$ roscd ompl/
ompl$ pwd
/opt/ros/noetic/share/ompl
ompl$ roscd chomp
roscd: No such package/stack 'chomp'
```
Since this is just a demonstration, I am not sure whether this error will influence my usage afterward

rhaschke, 2022-10-07:
What about just installing the missing ROS packages? See [here](http://repositories.ros.org/status_page/ros_noetic_default.html) for a list of all available packages.