moveit/panda_moveit_config #124: How to use Moveit on FR3

BolunDai0216, 2022-10-12:
Hi,

I was wondering if it would be possible to point me to some resources that show how to use moveit on a real robot. I did not see `franka_control.launch` used in the other launch files.

Thanks in advance!

rhaschke, 2022-10-13:
You use MoveIt in exactly the same fashion on a real robot, a Gazebo simulation, or the even simpler rviz-only simulation. Just tell MoveIt which robot controllers to use for actuation. A corresponding tutorial can be found here:
https://ros-planning.github.io/moveit_tutorials/doc/controller_configuration/controller_configuration_tutorial.html

rhaschke, 2022-10-13:
franka_control.launch is used in franka_control.launch of this package:
https://github.com/ros-planning/panda_moveit_config/blob/b5ce35571832b4165f67f5db51c69581354d2cee/launch/franka_control.launch#L7

closed 2022-10-13 by rhaschke