moveit/panda_moveit_config #128: remove `groud_plane` in gazebo

chauvinSimon, 2022-12-16:
By default, [`roslaunch panda_moveit_config demo_gazebo.launch`](https://github.com/ros-planning/panda_moveit_config/blob/noetic-devel/launch/demo_gazebo.launch) populates `Models` with two objects:
- the panda
- a `ground_plane`

Internally, `demo_gazebo.launch` launches [`franka_gazebo/launch/panda.launch`](https://github.com/frankaemika/franka_ros/blob/develop/franka_gazebo/launch/panda.launch) from `franka_gazebo`

As shown on the animation, the `ground_plane` acts as an obstacle and prevents the robot from reaching negative `z`

[image]

A solution is to **manually remove `ground_plane`, but this needs to be done each time gazebo is started**

My question: **is it possible to remove `ground_plane` permanently?**

I guess it must be **defined in some config file**
- Nevertheless **I could not find where**
- In [`stone.sdf`](https://github.com/frankaemika/franka_ros/blob/develop/franka_gazebo/world/stone.sdf), removing the section of `<uri>model://ground_plane</uri>` does not help.

rhaschke, 2022-12-17:
Loading `franka_gazebo/panda.launch` allows to specify your own Gazebo world model: https://github.com/rhaschke/franka_ros/blob/cartesian-impedance/franka_gazebo/launch/panda.launch#L8

Specifying `stone.sdf` with groundplane removed, works as expected:
`roslaunch franka_gazebo panda.launch world:=$PWD/src/robots/franka_ros/franka_gazebo/world/stone.sdf x:=-1`
[image]

The default world, Gazebo's `/usr/share/gazebo-11/worlds/empty.world` comprises the ground plane as well.

chauvinSimon, 2022-12-19:
Brilliant!
Thank you @rhaschke for the quick and complete answer!

closed 2022-12-19 by chauvinSimon