moveit/panda_moveit_config #131: Run stomp planner error "The 'stomp' configuration parameter was not found"

hlhfhmt, 2023-01-19:
I install moveit and stomp planner from source in Ubuntu 20. When i run the demo for the stomp planner with the command "roslaunch panda_moveit_config demo_stomp.launch ", i get the following error:
`[ERROR] [1674098279.924124491]: The 'stomp' configuration parameter was not found
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to initialize planning plugin
[move_group-5] process has died [pid 3939941, exit code -6, cmd /home/dell/ws_moveit/devel/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/dell/.ros/log/df160252-97a7-11ed-b47c-ebb3863f028e/move_group-5.log].
log file: /home/dell/.ros/log/df160252-97a7-11ed-b47c-ebb3863f028e/move_group-5*.log
[ INFO] [1674098280.451824028]: Stereo is NOT SUPPORTED
[ INFO] [1674098280.451920599]: OpenGL device: NVIDIA GeForce RTX 3090/PCIe/SSE2
[ INFO] [1674098280.451941955]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1674098283.856437647]: Loading robot model 'panda'...
[ INFO] [1674098283.959639331]: Starting planning scene monitor
[ INFO] [1674098283.961705455]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1674098284.367928116]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ INFO] [1674098289.374384433]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[ INFO] [1674098289.374592777]: Constructing new MoveGroup connection for group 'panda_arm' in namespace ''
[ERROR] [1674098319.388580579]: Unable to connect to move_group action server 'move_group' within allotted time (30s)
[ INFO] [1674098319.389808524]: Constructing new MoveGroup connection for group 'panda_arm' in namespace ''
[ERROR] [1674098349.402577899]: Unable to connect to move_group action server 'move_group' within allotted time (30s)
`
All planner config are right, any solutions to deal with that?

ChuGuoqing, 2023-04-22:
I got the same error. Have you solved the problem?

new2me321, 2023-06-08:
I am also facing the same error. Did anyone find the solution?

new2me321, 2023-06-08:
Hello, I found a solution to the issue. 
There is no need to define the planning group name in the file "stomp_planning_pipeline.launch.xml".  
~~group ns="$(arg arm_id)_arm"~~

Just delete all the lines with the "group"  tag in that file and replace them with 
`<rosparam command="load" file="$(find panda_moveit_config)/config/stomp_planning.yaml"/>`
[image]

However, you need to define the planning group directly in the 'stomp_planning.yaml' file

`stomp/panda_arm:`
`  group_name: panda_arm`
`  optimization:`
`    num_timesteps: 60`
`    num_iterations: 40`
   ` ...`

ChuGuoqing, 2023-06-09:
excellent work！！ Thank you very much 

824738078
***@***.***

 

 
 <rosparam command="load" file="$(find panda_moveit_config)/config/stomp_planning.yaml"/&gt;

Joinjiang, 2023-07-06:
but, it still have a another the same error show up? how to solve it

ravijo, 2023-07-30:
Thank you very much. I submitted the following pull request based on this thread.
https://github.com/ros-planning/panda_moveit_config/pull/137

The changes can be seen on the following commit:
https://github.com/ravijo/panda_moveit_config/commit/9a02d4df0ba97c9df5d1e06d490d053f02ff6229

ravijo, 2023-07-30:
@Joinjiang 

Until the pull request is approved and merged, you can pull the changes from the following repository:
https://github.com/ravijo/panda_moveit_config

closed 2023-07-30 by rhaschke

ravijo, 2023-07-31:
As for future readers, the latest commit https://github.com/ros-planning/panda_moveit_config/commit/058a848abef4e527b25b89f9d3d13ce716855bd8 by @rhaschke, proposes a much cleaner and better solution. Please use the [official repository](https://github.com/ros-planning/panda_moveit_config).

PR #138 (2023-07-30): Fix STOMP config: use extra stomp namespace
Fixes #131