moveit/panda_moveit_config #132: Something Wrong when following the tutorial "MoveIt Quickstart in RViz"

ChuGuoqing, 2023-04-04:
[image]
@v4hn I install the moveit from source following the office instument "https://moveit.ros.org/install/source/“（ubuntu 20.04，ROS Noetic）. When I follow the tutorial "MoveIt Quickstart in RViz", although I have make the robot collsion with itself, the links that are in collision doesn't  turn red(As shown in picture). What's wrong with this. Please help me.Thank you very much!

rhaschke, 2023-04-04:
This seems to be broken already in 1.1.6, the oldest release that easily compiles moveit_ros_vis against current 1.1.11.

rhaschke, 2023-04-08:
It turns out that the issue is not stemming from MoveIt, but from the new Panda URDF, which uses coarse collision-only models (named `*_sc`). These (and only these) are colored red but are not visible.
To handle such a config, we should think about coloring another link model instead. Maybe the first rigidly-connected parent link that has a visual?

ChuGuoqing, 2023-04-11:
> This seems to be broken already in 1.1.6, the oldest release that easily compiles moveit_ros_vis against current 1.1.11.

thank you very much