moveit/panda_moveit_config #133: ROS 2 Branch

peterdavidfagan, 2023-04-21:
Hi @rhaschke, 

Would a pull request for ROS 2 configuration files be welcomed? 

I also wished to ask, how are you deciding on what links to enable collisions for? I have been using MSA but I have encoutered some collisions while planning, I suspect due to the coarse collision geometries specified in the urdf of my robot (inherited from the default description files provided by Franka Emika).

rhaschke, 2023-04-21:
> Would a pull request for ROS 2 configuration files be welcomed?

Sure. I have created a rolling-devel branch.

> How are you deciding on what links to enable collisions for?

The Panda has a very specific URDF with its coarse collision geometries. This is not handled nicely by MSA's automated collision checking (essentially the coarse models always collide). Please inherit from the noetic-devel configuration and only apply required changes for ROS2.

peterdavidfagan, 2023-04-21:
Thanks @rhaschke.

> Please inherit from the noetic-devel configuration and only apply required changes for ROS2.

Will do.

130s, 2023-06-07:
I've started in https://github.com/ros-planning/panda_moveit_config/pull/136
@peterdavidfagan  If you have work-in-progress that you don't mind sharing, I'd like to see.

130s, 2023-06-07:
> I've started in #136 

#136 is ready for the review.

closed 2023-06-13 by rhaschke