moveit/panda_moveit_config #135: Run trajopt planner error "Exception while loading planner 'trajopt_interface/TrajOptPlanner'"

ChuGuoqing, 2023-04-22:
I install moveit and stomp planner from source in Ubuntu 20.04. When i run the demo for the trajopt planner with the command "roslaunch panda_moveit_config demo.launch  pipeline:=trajopt", i get the following error:

[ERROR] [1682185720.405746704]: Exception while loading planner 'trajopt_interface/TrajOptPlanner': According to the loaded plugin descriptions the class trajopt_interface/TrajOptPlanner with base class type planning_interface::PlannerManager does not exist. Declared types are  chomp_interface/CHOMPPlanner lerp_interface/LERPPlanner ompl_interface/OMPLPlanner pilz_industrial_motion_planner::CommandPlanner stomp_moveit/StompPlannerManager
Available plugins: chomp_interface/CHOMPPlanner, lerp_interface/LERPPlanner, ompl_interface/OMPLPlanner, pilz_industrial_motion_planner::CommandPlanner, stomp_moveit/StompPlannerManager

The complete prompt information is shown in "error.log". And I try to use ompl, chomp and pilz_industrial_motion_planner, they could run successfully. And I try to use stomp, I also get some similar error(same to #131 ). What' s wrong with this, could you help me? Thank you very much! @jspricke @rhaschke 
[error.log](https://github.com/ros-planning/panda_moveit_config/files/11301988/error.log)

rhaschke, 2023-04-23:
The error message states that the corresponding planner plugin cannot be found. Did you install the corresponding package?

ChuGuoqing, 2023-04-28:
I thought I have installed the corresponding package following the guide ‘‘https://moveit.ros.org/install/source/’’, and I could find the package in the corresponding folder(As shown below).
[image]

rhaschke, 2023-04-28:
Did you source your catkin workspace after building? What's the output of `echo ROS_PACKAGE_PATH`?

Noah-Lau, 2023-06-09:
Did you have any solution of this problem? @ChuGuoqing

Noah-Lau, 2023-06-09:
I got the same problem, and my  output of ` echo ROS_PACKAGE_PATH` shown as below:
/home/noah/ws_moveit/src/geometric_shapes:/home/noah/ws_moveit/src/moveit/moveit:/home/noah/ws_moveit/src/moveit_msgs:/home/noah/ws_moveit/src/moveit/moveit_planners/moveit_planners:/home/noah/ws_moveit/src/moveit/moveit_plugins/moveit_plugins:/home/noah/ws_moveit/src/moveit_resources/moveit_resources:/home/noah/ws_moveit/src/moveit_resources/dual_panda_moveit_config:/home/noah/ws_moveit/src/moveit_resources/fanuc_description:/home/noah/ws_moveit/src/moveit_resources/fanuc_moveit_config:/home/noah/ws_moveit/src/moveit/moveit_commander:/home/noah/ws_moveit/src/moveit_resources/panda_description:/home/noah/ws_moveit/src/moveit_resources/panda_moveit_config:/home/noah/ws_moveit/src/moveit_resources/pr2_description:/home/noah/ws_moveit/src/moveit_resources/prbt_moveit_config:/home/noah/ws_moveit/src/moveit_resources/prbt_pg70_support:/home/noah/ws_moveit/src/moveit_resources/prbt_support:/home/noah/ws_moveit/src/moveit/moveit_ros/moveit_ros:/home/noah/ws_moveit/src/moveit/moveit_runtime:/home/noah/ws_moveit/src/panda_moveit_config:/home/noah/ws_moveit/src/rviz_visual_tools:/home/noah/ws_moveit/src/srdfdom:/home/noah/ws_moveit/src/moveit/moveit_core:/home/noah/ws_moveit/src/moveit/moveit_planners/chomp/chomp_motion_planner:/home/noah/ws_moveit/src/moveit/moveit_planners/chomp/chomp_optimizer_adapter:/home/noah/ws_moveit/src/moveit_resources/prbt_ikfast_manipulator_plugin:/home/noah/ws_moveit/src/moveit/moveit_ros/occupancy_map_monitor:/home/noah/ws_moveit/src/moveit/moveit_ros/planning:/home/noah/ws_moveit/src/moveit/moveit_plugins/moveit_fake_controller_manager:/home/noah/ws_moveit/src/moveit/moveit_kinematics:/home/noah/ws_moveit/src/moveit/moveit_planners/ompl:/home/noah/ws_moveit/src/moveit/moveit_ros/move_group:/home/noah/ws_moveit/src/moveit/moveit_ros/manipulation:/home/noah/ws_moveit/src/moveit/moveit_ros/perception:/home/noah/ws_moveit/src/moveit/moveit_ros/robot_interaction:/home/noah/ws_moveit/src/moveit/moveit_ros/warehouse:/home/noah/ws_moveit/src/moveit/moveit_ros/benchmarks:/home/noah/ws_moveit/src/moveit/moveit_ros/planning_interface:/home/noah/ws_moveit/src/moveit/moveit_planners/chomp/chomp_interface:/home/noah/ws_moveit/src/moveit/moveit_ros/visualization:/home/noah/ws_moveit/src/moveit/moveit_ros/moveit_servo:/home/noah/ws_moveit/src/moveit/moveit_setup_assistant:/home/noah/ws_moveit/src/moveit/moveit_plugins/moveit_simple_controller_manager:/home/noah/ws_moveit/src/moveit/moveit_plugins/moveit_ros_control_interface:/home/noah/ws_moveit/src/moveit_visual_tools:/home/noah/ws_moveit/src/moveit_tutorials:/home/noah/ws_moveit/src/moveit/moveit_planners/pilz_industrial_motion_planner_testutils:/home/noah/ws_moveit/src/moveit/moveit_planners/pilz_industrial_motion_planner:/opt/ros/noetic/share.

where I can see the chomp and ompl plugins rather than trajopt. @rhaschke